open-sdr/openwifi-hw #1: Add compatibility for HackRF One

muellermartin, 2019-12-16:
I'd really like to see support for open hardware SDRs like the HackRF One (or the LimeSDR projects).

As far as I know, the HackRF One uses a CPLD instead of a FPGA. Unfortunaletely, I've no experience with FPGAs or alike therefore I would like know whether porting openwifi to the HackRF One is possible and what steps might be required for the port.

Some information about the HackRF One:
- hardware specifications: https://github.com/mossmann/hackrf/wiki/HackRF-One
CPLD
- hardware components: https://github.com/mossmann/hackrf/wiki/Hardware-Components
- CPLD firmware: https://github.com/mossmann/hackrf/tree/master/firmware/cpld

I'd be happy to help with whatever I can (e.g. testing).

JiaoXianjun, 2019-12-16:
Current openwifi fpga design is not highly optimized and needs around 27K Xilinx Kintex-7 LUTs. Hackrf’s cpld performance and volume are far below the openwifi requirement.

JiaoXianjun, 2020-03-29:
Now the smallest FPGA openwifi can run is Xilinx Zynq 7020 level.

closed 2020-03-29 by JiaoXianjun