open-sdr/openwifi-hw #3: Error when loading openofdm.tcl script from Vivado

frestuc, 2019-12-19:
Getting the following error while running the TCL script:

```
INFO: [BD 41-434] Could not find an IP with XCI file by name: system_axi_hdmi_core_0 
can't read "auto_set_param_list": no such variable
ERROR: [BD 41-1273] Error running init TCL procedure: can't read "auto_set_param_list": no such variable
    adi_auto_assign_device_spec Line 18
```

JiaoXianjun, 2019-12-20:
Which OS are you using? Do you do:
cd adi-hdl/library
git reset --hard 2018_r1
source $XILINX_DIR/Vivado/2017.4/settings64.sh
make
(Will take a while)

Successfully?

JiaoXianjun, 2019-12-20:
Would you please confirm that the issue is still there or not at your side?

frestuc, 2019-12-20:
The command is successful but the error is still there. Ubuntu 19.04.

JiaoXianjun, 2019-12-20:
What is the Vivado version? Could you paste full outputs of the command (making library and source the .tcl)

frestuc, 2019-12-21:
Vivado version is 2017.4.1

This is the vivado log: [https://pastebin.com/niJtQCLM](url)

JiaoXianjun, 2019-12-21:
So finally you can open the block diagram or cannot?

JiaoXianjun, 2019-12-21:
And, would you please try not to rename the openwifi.tcl? I notice that you rename it to rflearn_wifi.tcl

JiaoXianjun, 2019-12-21:
If you block diagram opening is successful, and synthesis&implementation run successfully. The message won't be a big issue .

frestuc, 2019-12-21:
Yes, S+I run successfully.

closed 2019-12-21 by frestuc