open-sdr/openwifi-hw #4: running openwifi in ZC702 or zedboard

eljuligallego, 2019-12-21:
Hi I would like to evaluate this project but since the target is the zc706 board with a zynq xc045 device a  Vivado license is needed and it's not free.

I tried to modify the target to the ZC702 eval board with the same FPGA than the Zedboard the xc7z020clg484 device by simply repurpose the goal device in tcl project and IP cores but after compile the design you provide the implementation is not possible due not enough resources. 

Here you have the utilization report from Vivado syntesys:

[image]

The resources exceed in SLUTs and DSPs. The first problem is about 13200 Slice LUTs that don't know how to reduce the design to achieve the goal since it's a big difference.

The DSPs maybe is easier to achieve, the 43 DSPs could be reduced by deleting the dds inside the ad9361(40 DSPs blocks) and re purpose other instance to use logic cells instead of DSPs.

I read in your road map this board will be compatible but seem a project redesign is needed unless i'm wrong in my test.

My desired goal would be the pluto board but I guess this will not be possible due the enormous lack of logic cells needed in this implementation.¿any approach to make it suitable for pluto board or zedboard?.

JiaoXianjun, 2019-12-21:
Indeed pluto is too hard.

I think zedboard is possible, because it has z7020 FPGA. 7020 FPGA is on our oradmap (adrv9364-z7020 has the same 7020 FPGA as zedboard).

Some possible simplification:
1. delete those useless module in the ADI reference design, such as HDMI related, audio related, etc.
2. shrink some FIFO size in the openwifi design. 
3. shrink the openwifi DDC (mixer+FIR). Currently it supports multiple 20M and 2M channels. It can be downsized to single 20M channel.
we are working on above 2 points. There will be commit soon.

frestuc, 2020-02-26:
Hi Xianjun,

Do you think it is easy(ish) to port OpenWiFi on ZCU111? 
Are there any effort from your group in this regard?

Best,
Frank

JiaoXianjun, 2020-02-26:
I have been working on this thread. Porting to zed and 702 have been done. adrv9361 and adrv9364 SOM also have been done! They work perfectly. 
We will have release later this week or early next week!

frestuc, 2020-02-26:
I am talking about ZCU111, not the ZC702.

JiaoXianjun, 2020-02-26:
In principle newer or bigger FPGA would be easier for porting.

closed 2020-03-06 by JiaoXianjun

JiaoXianjun, 2020-07-31:
@eljuligallego 

Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email (if you could also introduce yourself bit, that would be perfect). We might send out some questions to listen for user feed feedback.

Thanks.