open-sdr/openwifi-hw #5: ip openofdm_rx_0 is locked

jinsanko, 2019-12-26:
Please check this. There is no files in /zc706_fmcs2/ip/openofdm_rx after the make step.

Thank.

Jinsan Ko

weiliu1011, 2019-12-26:
Hi Jinsan Ko,
Do you have the Xilinx viterbi decoder license? If not it is going to appear as locked and cannot generate output products. You can apply an evaluation license free of cost.
Wei

closed 2019-12-26 by jinsanko

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email (if you could also introduce yourself bit, that would be perfect). We might send out some questions to listen for user feed feedback.

Thanks.