open-sdr/openwifi-hw #6: Questions on DDC/DUC design

frestuc, 2020-01-18:
Hi,

I was going through the design code and was wondering what is the purpose of the mixer in the DDC module in `rx_intf`. After you decimate the 40MHz band signal coming from the ADC to 20MHz, why do you further need a mixer? The signal should be centered to baseband, right? 

By looking at the `sdr.c` driver, it seems that you are setting the mixer configurations to two different values for TX and RX chains. May I ask why and what do these configs represent?

```
    priv->rx_intf_cfg = RX_INTF_BW_20MHZ_AT_0MHZ_ANT0;
    priv->tx_intf_cfg = TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1;
```

Last question -- by looking at `src.c`, I can see that you do not set the RF bandwidth to 40MHz manually -- may I ask why? Is it the default value for AD9361, and where can I check that this is actually the case?

Thanks a lot!

JiaoXianjun, 2020-01-18:
ddc_bank_core.bd, which is used by ddc_bank_core_intf.v, is real ddc.
mixer firstly. 40Msps sampling rate won't change. Then an FIR is used for decimation. After this, we have 20Msps.

You should search priv->rx_intf_cfg in whole source files, and to see which function (finally which module) is using it. Different DDC and DUC setting can have AD9361 tx lo is different from rx lo. In this way, self-interference is minimized. Please see ad9361_rf_set_channel in sdr.c to understand the configuration.

When wgd.sh is called, it calls rf_inti.sh to do AD9361 initial setting. Depends on board, rf_init.sh is now rf_init_zc706_fmcs2.sh or rf_init_adrv9361z7035.sh.

I will write a section in document next week.

frestuc, 2020-01-18:
Ok -- you are setting the TX LO to 5.25 (5.24 + 10MHz) GHz and then using the DUC to bring the bb signal from 0 to -10MHz to make the two LOs work at different frequencies.

Another question -- why do you set the rf bandwidth of the RX to 17.5MHz and of the TX to 37.5MHz?

```
echo 17500000 >  in_voltage_rf_bandwidth
echo 37500000 >  out_voltage_rf_bandwidth 
```

JiaoXianjun, 2020-01-18:
Because rx ddc digital mixer centred at 0MHz, but tx ddc cantered at 10MHz offset. However, the front end bandwidth for both rx and tx is 40MHz. To cover the same 20MHz region for tx and rx, the setting has to be like that.

Using 17.5 not 20 because not all 64 subcarriers are used.

frestuc, 2020-01-18:
Yep, makes sense. Thanks.

JiaoXianjun, 2020-01-19:
Hello, the related document has been updated: https://github.com/open-sdr/openwifi/blob/master/doc/README.md

catkira, 2022-10-22:
@JiaoXianjun is the tx offet still used with the new LO control method described [here](https://arxiv.org/pdf/2204.07354.pdf)?

JiaoXianjun, 2022-10-24:
> @JiaoXianjun is the tx offet still used with the new LO control method described [here](https://arxiv.org/pdf/2204.07354.pdf)?

https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Analog-and-digital-frequency-design

catkira, 2022-10-24:
ok that looks like the answer is "no" which makes sense.