open-sdr/openwifi-hw #10: TCL procedure: unexpected "," outside function argument list

gmartina, 2020-05-13:
Hi,
In this step:
Open Block Design
Tools --> Report --> Report IP Status

I get the following error:
```
 [BD 41-1273] Error running post_config_ip TCL procedure: unexpected "," outside function argument list
    ::xilinx.com_ip_zynq_ultra_ps_e_3.2::post_config_ip Line 47

```
Any idea of how to solve it?

best regards

JiaoXianjun, 2020-05-13:
which OS and version? Which board project are you working on?

closed 2020-06-12 by JiaoXianjun

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email (if you could also introduce yourself bit, that would be perfect). We might send out some questions to listen for user feed feedback.

Thanks.