open-sdr/openwifi-hw #11: Simulation of openofdm_rx

hyanki, 2020-06-09:
Hi 
Its really great.
I want to know that how can i simulate these designs?
Regards
hyanki

JiaoXianjun, 2020-06-09:
Yes it can be simulated.
I simulated it in Vivado via the testbench: https://github.com/open-sdr/openofdm/blob/abbe9ecde9e08267386fd7263707c9a9557df4c3/verilog/dot11_tb.v

hyanki, 2020-06-09:
Hi Jiao Xianjun
thanks for your response. 
Is it possible to simulate icarus verilog?

Regards
hyanki

JiaoXianjun, 2020-06-09:
I think it is possible.

hyanki, 2020-06-10:
Hi Jiao Xianjun
Do you have any idea how it can be done because i have not worked on icarus.
Regards
hyanki

JiaoXianjun, 2020-06-10:
I also do not have experience on that tool. But I guess the procedure should be similar. Add all source files into the project then simulate the top level test bench.

hyanki, 2020-06-10:
I have tried but it gives error that modules are missing. any way in vivado how you have feeded input data to rf modulde and how you have set register setting i mean min_plateu etc.

mmehari, 2020-06-10:
Hello,
The original openofdm module supports icarus verilog simulation and you can use the following commit (https://github.com/jhshi/openofdm/tree/2f0e0ba95385affa426dc11eba6effab37b8778b) for your test.

This, however, does not include our development but if you need that, you have to use vivado simulation as @JiaoXianjun pointed you.

hyanki, 2020-06-11:
Hi
problem is resolved now
thanks a lot

closed 2020-06-11 by hyanki