open-sdr/openwifi-hw #12: Soft decoding in FEC block

hyanki, 2020-06-11:
Hi 
i have doubt in soft decoding applied in FEC.
output of de-modulator in 64QAM case, output bit width is six for hard decoding as well as soft bits.
How is it possible, I guess no of soft bits will be more than no of hard bits?
Regards
Hyanki

mmehari, 2020-06-11:
Hello Hyanki,
The goal of soft decoding to limit the quantization error of IQ demodulation so that the Viterbi decoder will make a better estimate during the decoding process.

Furthermore, at any given constellation point and modulation scheme (i.e. BPSK, QPSK, 16QAM, 64QAM), only a single bit per phase (I.e in-phase, quadrature) will experience quantization error. You can check it yourself on the constellation diagram (https://openofdm.readthedocs.io/en/latest/_images/mod.png). 

This means, we only need to save the soft decoding information for two bits and the position within the constellation diagram. Now the Viterbi decoder from Xilinx by default uses 3-bit width for soft-decoding representation and that makes the total bits to 2*3 = 6. 

So for your question, yes the number of bits in soft and hard decoding is the same but they have different meaning

hyanki, 2020-06-11:
Hi mmehri
In 64QAM we are packing 6 bit in one symbol then modulate. In receiver we demodulate it and feeded to viterbi but   i think viterbi is giving one bit output.
Here how these packed bits are reterved using viterbi?
Regards
hyanki

mmehari, 2020-06-11:
In hard decoding mode, the Viterbi decoder accepts two bits and produces 1 bit. Whereas in the soft decoding mode, the Viterbi decoder accepts 6 bits (2*3) and produces 1 bit.

Inside openofdm, the Viterbi decoder from Xilinx was always enabled to work in soft decoding mode but the original author @jhshi has not made the module to take advantage of this. That means, the module was working in hard decoding mode and that is what we have updated in our development.

As to the principle of working of the Viterbi decoder, you can look at the following youtube movies
https://www.youtube.com/watch?v=kRIfpmiMCpU
https://www.youtube.com/watch?v=dKIf6mQUfnY

hyanki, 2020-06-11:
Hi michael
As FEC adds redundancy bits.  lets take 1/2 rate convolution so encoder add
1 parity bit and when these two encoded bits fedded to viterbi it will
remove parity and give one original bit. Its clear in case of QPSK.
But when i take case of 16QAM and convolution is  2/3 rate. Means 1 parity
bit for 2 bits. So after viterbi decoding how i will reteriave two original
bit.
Regards
Hyanki

mmehari, 2020-06-11:
Where did you get the information that 16QAM uses convolution rate 2/3?`

hyanki, 2020-06-12:
Hi michael
I have assumed. Still i have doubt how it will happen. I think i could not
communicted my point.
 in convolution encoder
We are feeding single bit and getting two bit output for 1/2 rate. In
encoding for 16 QAM we pack 4 bit in one symbol( two parity and two info
bit).
In receiver we get one symble at a time means it is having 4 bit
information (2bit information + 2bit parity).
Now my doubt is that how  viterbi is decoding 2 bit. Are you doing parallel
precessing here or used two viterbi block parallellely or viterbi is output
two bits serially.
Regards
Hyanki

mmehari, 2020-06-12:
First of all, we haven't implemented the Viterbi decoder, except using the IP core from Xilinx.

Second, the Viterbi decoder works on a sequence of inputs before producing an output, rather than 4bits in the case of 16 QAM. The operation of the Viterbi decoder does not depend on the modulation schemes. All it cares about is the bits.

So I would advise you to watch the youtube videos I have attached previously.

hyanki, 2020-06-12:
Thanks for your support.
I was just trying to steps involved in decoding process.
Thanks again.
Regards
Hyanki

closed 2020-06-12 by JiaoXianjun