open-sdr/openwifi-hw #17: Problems when simulating openofdm_rx

frestuc, 2020-07-21:
I get these errors when trying to simulate openofdm_rx with Vivado 2018.3:

```INFO: [USF-XSim-97] Finding global include files...
INFO: [USF-XSim-98] Fetching design files from 'sim_1'...
INFO: [USF-XSim-2] XSim::Compile design
INFO: [USF-XSim-61] Executing 'COMPILE and ANALYZE' step in '/home/frank/Documents/my_openwifi/openwifi-hw/boards/zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.tmp/openofdm_rx_v1_0_project/openofdm_rx_v1_0_project.sim/sim_1/behav/xsim'
xvlog --incr --relax -prj dot11_tb_vlog.prj
ERROR: [XSIM 43-4307] Invalid entry in the project file, expecting a file name but specified path '../../../../../../../../../ip_repo/common/openofdm_rx/src' represents a directory name.
ERROR: [XSIM 43-3217] dot11_tb_vlog.prj (line 3): Incorrect project file syntax. Correct syntax is one of: vhdl <worklib> <file>, verilog <worklib> <file> [<file> ...] [[-d <macro>] ...] [[-i <include>] ...], or NOSORT. Presence of NOSORT on a line of its own disables file order sorting.
xvhdl --incr --relax -prj dot11_tb_vhdl.prj
```
Any suggestion? Thanks

mmehari, 2020-07-21:
Can you be more specific on the steps you took before coming to this error?
It will be helpful to re-create the problem from our side.

frestuc, 2020-07-21:
I just hit the "Run Simulation" button on the side panel.

frestuc, 2020-07-21:
This is the code of dot11_tb_vlog.prj, apparently it complains that a folder is included?

```# compile verilog/system verilog design source files
verilog xil_defaultlib  --include "../../../../../../../../../ip_repo/common/openofdm_rx/src" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/deinter_lut/sim/deinter_lut.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/div_gen_xlslice_0_0/sim/div_gen_xlslice_0_0.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/atan_lut/sim/atan_lut.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/rot_lut/sim/rot_lut.v" \

verilog   --include "../../../../../../../../../ip_repo/common/openofdm_rx/src" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/bits_to_bytes.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/calc_mean.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/complex_mult.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/complex_to_mag.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/complex_to_mag_sq.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/crc32.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/deinterleave.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/delayT.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/delay_sample.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/demodulate.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/descramble.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/div_gen.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/divider.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/dot11.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/equalizer.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/ht_sig_crc.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/moving_avg.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/ofdm_decoder.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/phase.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/ram_2port.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/rotate.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/stage_mult.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/sync_long.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/sync_short.v" \
"../../../../../../../../../ip_repo/common/openofdm_rx/src/dot11_tb.v" \

# compile glbl module
verilog xil_defaultlib "glbl.v"

# Do not sort compile order
nosort
```

JiaoXianjun, 2020-07-21:
The project name seems strange. Would you please describe the steps to generate the vivado project.

frestuc, 2020-07-21:
It's the automated project created by "Edit in IP Packager"...

JiaoXianjun, 2020-07-21:
You need to create the project by following “openofdm_rx” section under “modify ip cores” in openwifi-hw readme. the openwifi-hw should be under openwifi directory (by submodule command). Then openofdm_rx should be under openwifi-hw/ip by submodule command. 

Above method and directory structure could simulate openofdm_rx smoothly.

frestuc, 2020-07-21:
Thanks, it worked.

closed 2020-07-21 by frestuc