open-sdr/openwifi-hw #18: Data format of IQ

hyanki, 2020-07-23:
Hi
I want to know that what is data format of IQ in 16 bit in receive  side?
In demodulater.v why scaling is taken 1024 though input is in 16 bit?
Regards
J S Hyanki

JiaoXianjun, 2020-07-31:
You can read document of openofdm: https://openofdm.readthedocs.io/en/latest/

hyanki, 2020-07-31:
Hi jiao
Basically in demodulate.v , input is 16 bit
but for demaping max value is taken 1024 which is 10 bit. Why?
Regards
J S Hyanki

JiaoXianjun, 2020-07-31:
because the signal is scaled to that range by the processing steps.

hyanki, 2020-07-31:
You mean output of FFT block.

JiaoXianjun, 2020-07-31:
Could be. And also equalization.

hyanki, 2020-07-31:
So maximum value coming from fft block is 1024.

JiaoXianjun, 2020-07-31:
from euqalization module. Fft output is not fed to demodulate module directly. There is a equalisation module before the demodulation. 

1024 is just a reference amplitude to decide the  decision threshold inside demodulation.

hyanki, 2020-07-31:
Ok.
I want to know that how much doppler can cope this ref design?

JiaoXianjun, 2020-07-31:
We never test this. I think it should be like several 10KHz.

hyanki, 2020-07-31:
Ok
Thanks for your support.
Regards

closed 2020-07-31 by JiaoXianjun

JiaoXianjun, 2020-07-31:
Hello,

Would you please tell us your email? If you could also introduce yourself bit (student? Company? University?), that would be perfect. We might send out some questions to listen for user feed feedback.

Thanks.

hyanki, 2020-08-01:
Hi
My mail id is hyankijitu@gmail.com
Regards
J S Hyanki