open-sdr/openwifi-hw #26: system module

fatihilig-fpga, 2020-11-30:
First of all thank you for your help.
I couldn't find the system.v sub-module which is attached into dropbox link shown below. 
https://www.dropbox.com/s/wmvgghgtb90bjm4/issuePic.png?dl=0

[https://github.com/open-sdr/openwifi-hw/blob/master/boards/adrv9361z7035/src/system_wrapper.v](url)
Have a great day.

JiaoXianjun, 2020-12-04:
Hello,

Yes we can help. But firstly would you please send me (xianjun.jiao@ugent.be) information about you? Student? (MS? PhD?), or employee of which company? Thanks!

suokea, 2021-01-28:
您好，请问您有没有关于xilinx SDK的代码，麻烦能不能让我参考一下

suokea, 2021-01-28:
Hello!
Do you have any Xilinx SDK code for me to refer to? Thank you very much

JiaoXianjun, 2021-01-28:
> Hello!
> Do you have any Xilinx SDK code for me to refer to? Thank you very much

Hello,

Yes we can help. But firstly would you please send me (xianjun.jiao@ugent.be) information about you? Student? (MS? PhD?), or employee of which company? Thanks!

JiaoXianjun, 2021-01-29:
The SDK files are in openwifi-hw/boards/zcu102_fmcs2/sdk/

Please Check the sdk_update.sh in README:

```
In Vivado:
Open Block Design
Tools --> Report --> Report IP Status
Generate Bitstream
(Will take a while)
File --> Export --> Export Hardware... --> Include bitstream --> OK
File --> Launch SDK --> OK, then close SDK
In Linux:
cd openwifi-hw/boards
./sdk_update.sh board_name
git commit -a -m "new fpga img for openwifi (or comments you want to make)"
git push
(Above make sure you can pull this new FPGA from openwifi submodule directory: openwifi-hw)
```

suokea, 2021-01-29:
Hello!
Sorry to disturb you again. I found the openwifi HW code on GitHub. After downloading it into a zip file, what's the reason for the decompression failure?

（您好！
抱歉，再次打扰。我在GITHUB上找到了openwifi-hw的代码，下载成ZIP文件后，解压失败是什么什么原因呢？）

 

 
 In Vivado: Open Block Design Tools --&gt; Report --&gt; Report IP Status Generate Bitstream (Will take a while) File --&gt; Export --&gt; Export Hardware... --&gt; Include bitstream --&gt; OK File --&gt; Launch SDK --&gt; OK, then close SDK In Linux: cd openwifi-hw/boards ./sdk_update.sh board_name git commit -a -m "new fpga img for openwifi (or comments you want to make)" git push (Above make sure you can pull this new FPGA from openwifi submodule directory: openwifi-hw)

JiaoXianjun, 2021-01-29:
The zip file is not created by us. It is created by github when we publish a release.
I suggest you download again. Or use git clone command.

closed 2021-01-29 by JiaoXianjun