open-sdr/openwifi-hw #34: after Updating  latest  FPGA bitstream openwifi get fail to initiation

kelvin820, 2021-04-21:
Hi,
Following **Quick Start** instructions I can see openwifi AP on my phone and get connected successfully. However, after updating latest FPGA bitestream  following **Update FPGA** instructions. My board always get hanged. 

I use 

Ubuntu 18.04.5 LTS
Vivado 2018.3 without license
Analog Device ADRV9361-Z7035

I don't see any error message during BOOT.BIN generation. After I replace BOOT.BIN on SD card / power cycle and try to inite  ~/openwifi/fosdem-11ag.sh

...
...
...

initiationsdr,sdr openwifi_dev_probe: test_mode 0
sdr,sdr openwifi_dev_probe: rfkill radio on
sdr,sdr openwifi_dev_probe: band_2GHz.n_channels 14 n_bitrates 12 band_5GHz.n_channels 11 n_bitrates 12
sdr,sdr openwifi_dev_probe: mac_addr 66:55:44:33:22:d5
ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
sdr,sdr openwifi_dev_probe: ieee80211_register_hw 0
ieee80211 phy0: hwaddr 6655443322d5, ZYNQ + ad9361
sdr,sdr openwifi_rfkill_init: wireless switch is on
sdr,sdr fpga-axi@0:sdr sdr0: renamed from wlan0
stop: Unknown instance: 
isc-dhcp-server start/running, process 2263
Configuration file: hostapd-openwifi-11ag.conf
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface sdr0 with hwaddr 66:55:44:33:22:d5 and ssid "openwifi"
**NOT COMPLETE THE WHOLE PROCESS  HANGED HERE**
------------------------------

board get hanged every time without complete the whole initiation process. So I cannot see openwifi AP nor use it.

Would you please help to advised which part may I go wrong? Thank for your great help.

JiaoXianjun, 2021-04-21:
Hello,

Thanks for reaching us. We could take a look. Before that, would you please let us know whom we are helping? Could you send your information (name, organization/university/company, etc) from your organization email to my email xianjun.jiao@ugent.be?

Thanks!

JiaoXianjun, 2021-04-26:
Hello,

Can you paste the command the output here when you do " Update FPGA instructions"?

kelvin820, 2021-04-26:
I've stuck to the readme.md 
##  1. Export variables
```
export XILINX_DIR=/tools/Xilinx
export OPENWIFI_DIR=~/FpgaPlatform/openwifi
export BOARD_NAME=adrv9364z7020
```
###  2. Generate BOOT.BIN
```
$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME
```

###  3. Upload BOOT.BIN to board
```
scp $OPENWIFI_DIR/kernel_boot/boards/$BOARD_NAME/output_boot_bin/BOOT.BIN root@192.168.10.122:
```

###  4. Replace BOOT.BIN
```
mount /dev/mmcblk0p1 /mnt
cp ~/BOOT.BIN /mnt
umount /mnt
```

JiaoXianjun, 2021-04-26:
Are you doing these right after you flash the sdcard image into your sdcard? You never modify anything? What is the output of these two commands:
```
cd openwifi
git log
cd openwifi/openwifi-hw
git log
```
What are the output of all your command listed in your previous comment?

kelvin820, 2021-04-26:
I don't modify anything.  Could this been the problem of my Vivado? 

Here is git log of openwifi
```
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi$ git log
commit e1a9c183bc2fed27732a600c2c84c6e828db16b9 (HEAD -> master, origin/master, origin/HEAD)
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Wed Apr 7 10:00:20 2021 +0200

    update jpg in webserver

commit 09d381a0931f47f1f78f374e7490f3b2ac570fcd
Author: Jiao Xianjun <xianjun.jiao@ugent.be>
Date:   Tue Apr 6 18:04:47 2021 +0200

    Update README.md

commit b5a0a70cb645a339974e27d357f2bcc6a5861fd1
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Tue Apr 6 10:06:49 2021 +0200

    doc/README.md udpate

...
..
.
```

And Here is the git log of /openwifi/openwifi-hw
```
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi$ cd openwifi-hw/
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi/openwifi-hw$ git log
commit f8eb553a71f28dcbb2764ce326d577c0d352eeeb (HEAD -> master, origin/master, origin/HEAD)
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Mon Apr 5 21:34:09 2021 +0200

    udpate readme accordingly

commit 4b60853efa5121716a216105fb8ae9f9d954896c
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Mon Apr 5 21:32:12 2021 +0200

    udpate fpga bitstream

commit c7007da34a2e3509bb44cfa0f7b23189795caad3
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Mon Apr 5 21:30:45 2021 +0200

    udpate ip_repo

commit 1d8d3d9a56b91ee75e68e2324048f8d7568e5330
Author: Xianjun Jiao <xianjun.jiao@ugent.be>
Date:   Mon Apr 5 21:26:56 2021 +0200

    in xpu.v slv_reg19 and slv_reg8 are not twistted anymore. slv_reg6 is added to assist the register map in xpu more clear. separate registers for different purpose. separate registers for dynamic and static configurations in driver (sdr.c).
...
..
.

```

kelvin820, 2021-04-26:
Here are the terminal output log of  command ```$OPENWIFI_DIR/user_space/get_fpga.sh $OPENWIFI_DIR```

```
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi$ $OPENWIFI_DIR/user_space/get_fpga.sh $OPENWIFI_DIR
$OPENWIFI_DIR is found!
+ cd /home/xw/FpgaPlatform/openwifi//
+ git submodule init openwifi-hw
+ git submodule update openwifi-hw
+ cd openwifi-hw
+ git checkout master
Already on 'master'
Your branch is up to date with 'origin/master'.
+ git pull
Already up to date.
+ cd /home/xw/FpgaPlatform/openwifi

```

Here are the terminal output log of  command ```$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME```

```
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi$ $OPENWIFI_DIR/user_space/get_fpga.sh $OPENWIFI_DIR
$OPENWIFI_DIR is found!
+ cd /home/xw/FpgaPlatform/openwifi//
+ git submodule init openwifi-hw
+ git submodule update openwifi-hw
+ cd openwifi-hw
+ git checkout master
Already on 'master'
Your branch is up to date with 'origin/master'.
+ git pull
Already up to date.
+ cd /home/xw/FpgaPlatform/openwifi
xw@xw-HP-EliteBook-840-G2:~/FpgaPlatform/openwifi$ $OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME
$OPENWIFI_DIR is found!
$XILINX_DIR is found!
$BOARD_NAME is found!
+ source /tools/Xilinx//SDK/2018.3/settings64.sh
++ source /tools/Xilinx/Vivado/2018.3/.settings64-Vivado.sh
+++ export XILINX_VIVADO=/tools/Xilinx/Vivado/2018.3
+++ XILINX_VIVADO=/tools/Xilinx/Vivado/2018.3
+++ '[' -n /tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
+++ export PATH=/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ PATH=/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++ source /tools/Xilinx/DocNav/.settings64-DocNav.sh
+++ '[' -n /tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
+++ export PATH=/tools/Xilinx/DocNav:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ PATH=/tools/Xilinx/DocNav:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++ source /tools/Xilinx/SDK/2018.3/.settings64-SDK_Core_Tools.sh
+++ '[' -n /tools/Xilinx/DocNav:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
+++ export PATH=/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/DocNav:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ PATH=/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/DocNav:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/SDK/2018.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+ cd /home/xw/FpgaPlatform/openwifi//kernel_boot
+ ./build_boot_bin.sh adrv9361z7035
$BOARD_NAME is found!
+ HDF_FILE=../openwifi-hw/boards/adrv9361z7035/sdk/system_top_hw_platform_0/system.hdf
+ UBOOT_FILE=./boards/adrv9361z7035/u-boot.elf
+ BUILD_DIR=./boards/adrv9361z7035/build_boot_bin
+ OUTPUT_DIR=./boards/adrv9361z7035/output_boot_bin
+ echo ../openwifi-hw/boards/adrv9361z7035/sdk/system_top_hw_platform_0/system.hdf
+ grep -q .hdf
+ echo ./boards/adrv9361z7035/u-boot.elf
+ grep -q -e .elf -e uboot
+ '[' '!' -f ../openwifi-hw/boards/adrv9361z7035/sdk/system_top_hw_platform_0/system.hdf ']'
+ '[' '!' -f ./boards/adrv9361z7035/u-boot.elf ']'
+ command -v xsdk
+ command -v bootgen
+ rm -Rf ./boards/adrv9361z7035/build_boot_bin ./boards/adrv9361z7035/output_boot_bin
+ mkdir -p ./boards/adrv9361z7035/output_boot_bin
+ mkdir -p ./boards/adrv9361z7035/build_boot_bin
[... 232 lines cut ...]
 
[... cut ...]
+ cp ./boards/adrv9361z7035/build_boot_bin/build/sdk/fsbl/Release/fsbl.elf ./boards/adrv9361z7035/output_boot_bin/fsbl.elf
+ cp ./boards/adrv9361z7035/build_boot_bin/build/sdk/hw_0/system_top.bit ./boards/adrv9361z7035/output_boot_bin/system_top.bit
+ cd ./boards/adrv9361z7035/output_boot_bin
+ bootgen -arch zynq -image zynq.bif -o BOOT.BIN -w

****** Xilinx Bootgen v2018.3
  **** Build date : Dec  6 2018-23:41:49
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

+ rm ./boards/adrv9361z7035/build_boot_bin -rf
+ cd /home/xw/FpgaPlatform/openwifi

```

JiaoXianjun, 2021-04-26:
So you did all the operation right after you flash our image file into your SD card without any other modification, right?

JiaoXianjun, 2021-04-26:
I just repeated your step, and everything work well.

Some suggestions:

1.
on board when you run:
```
. Replace BOOT.BIN
mount /dev/mmcblk0p1 /mnt
cp ~/BOOT.BIN /mnt
umount /mnt
```
could be replaced by:
```
mount /dev/mmcblk0p1 /mnt
cd /mnt
sync
cp ~/BOOT.BIN /mnt -rf
sync
ls
cd
umount /mnt
```

2.
Use another SD card (maybe there are some bad sectors on your current SD card)

kelvin820, 2021-04-27:
> So you did all the operation right after you flash our image file into your SD card without any other modification, right?
> Without your operations, the orginal image can be used without any issue?

closed 2021-04-27 by kelvin820