open-sdr/openwifi-hw #35: Cannot build adrv9361z7035 board FPGA

kelvin820, 2021-04-23:
Hello,
I've tried to build adrv9361z7035 FPGA bit file from  scretch but fail. Fail at ```prepare_adi_board_ip.sh``` process. I got the error message as below. Could you help to advise  what may go wrong? BTW, I've tried all the board supported. Only **zed_fmcs2** , **adrv9364z7020** , **zc702_fmcs** could successfully complete the whole script process.

**vivado.log**
```
-----------------------------------------------------------
# Vivado v2018.3 (64-bit)
# SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
# IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
# Start of session at: Fri Apr 23 10:35:46 2021
# Process ID: 11123
# Current directory: /home/xw/trial/openwifi-hw/adi-hdl/projects/adrv9361z7035/ccbob_lvds
# Command line: vivado -mode batch -source system_project.tcl
# Log file: /home/xw/trial/openwifi-hw/adi-hdl/projects/adrv9361z7035/ccbob_lvds/vivado.log
# Journal file: /home/xw/trial/openwifi-hw/adi-hdl/projects/adrv9361z7035/ccbob_lvds/vivado.jou
#-----------------------------------------------------------
source system_project.tcl
# source ../../scripts/adi_env.tcl
....
..
.
# set p_device "xc7z035ifbg676-2L"
# adi_project adrv9361z7035_ccbob_lvds
WARNING: [Device 21-436] No parts matched 'xc7z035ifbg676-2L'
ERROR: [Coretcl 2-106] Specified part could not be found.
INFO: [Common 17-206] Exiting Vivado at Fri Apr 23 10:35:52 2021...
```

weiliu1011, 2021-04-23:
It looks like you don't have the device required in your vivado, can you follow [this link](https://forums.xilinx.com/t5/Vivado-TCL-Community/Coretcl-2-106-Specified-part-could-not-be-found/td-p/676020) to add the device? Apart from this you also need to have license for those devices in order to generate bitstream.

JiaoXianjun, 2021-04-26:
For FPGA larger than 7020, you need Vivado license to compile build the FPGA bitstream. These three "zed_fmcs2 , adrv9364z7020 , zc702_fmcs" all have 7020 FPGA, so they are small enough to go without Vivado license.

kelvin820, 2021-04-26:
@weiliu1011 I'll give a try and get back here.

kelvin820, 2021-04-27:
> It looks like you don't have the device required in your vivado, can you follow [this [...]

Thanks @weiliu1011 @JiaoXianjun I don't expect I should re-install vivado to make Vivado HL activated. Now I can build pass.

closed 2021-04-27 by kelvin820