open-sdr/openwifi-hw #36: build open-wifi

andrew712, 2021-06-22:
Hi, recently I began to experience problems with the build of the openwifi-hw project, I use vivado 2018.3 build configuration zc706 fmcs2, this problem did not arise before, I tried several times to download the repository but it didn’t help, can you tell me what this problem might be related to?

 [IP_Flow 19-3188] Error occurred while initializing 'system_sys_audio_clkgen_0'
Tcl error in update procedure while setting value 'MMCM' on the parameter 'PRIMITIVE'. FALSE

 [IP_Flow 19-3428] Failed to create Customization object system_sys_audio_clkgen_0
 [BD 41-1712] Create IP failed with errors
 [BD 41-595] Failed to add BD cell <sys_audio_clkgen>
 [IP_Flow 19-973] Failed to create IP instance 'system_sys_audio_clkgen_0'. Error during customization.

JiaoXianjun, 2021-06-22:
I just tried zc706_fmcs2, and don't have any issue.

Can you tell me what you have executed after you do a fresh git clone of openwifi-hw? And the error occurs in which step?

andrew712, 2021-06-22:
mars@mars-pc:~/open_wifi_5/openwifi-hw$ export BOARD_NAME=zc706_fmcs2mars@mars-pc:~/open_wifi_5/openwifi-hw$ export XILINX_DIR=/home/mars/Xilinx/Vivado/
mars@mars-pc:~/open_wifi_5/openwifi-hw$ ./prepare_adi_lib.sh $XILINX_DIR$XILINX_DIR is found!
++ git submodule init adi-hdl
Submodule 'adi-hdl' (https://github.com/analogdevicesinc/hdl.git) registered for path 'adi-hdl'
++ git submodule update adi-hdl
Cloning into 'adi-hdl'...
remote: Enumerating objects: 59191, done.
remote: Counting objects: 100% (366/366), done.
remote: Compressing objects: 100% (179/179), done.
remote: Total 59191 (delta 214), reused 288 (delta 186), pack-reused 58825
Receiving objects: 100% (59191/59191), 15.70 MiB | 189.00 KiB/s, done.

Resolving deltas: 100% (42379/42379), done.
Checking connectivity... done.
Submodule path 'adi-hdl': checked out 'f61d9707eb0a62533efd6facab59ab2444da94c9'
++ cd ./adi-hdl/library
++ git reset --hard f61d9707eb0a62533efd6facab59ab2444da94c9
HEAD is now at f61d970 ad_fmclidar1_ebz: Update the IO constraints to revB
++ cd /home/mars/open_wifi_5/openwifi-hw
mars@mars-pc:~/open_wifi_5/openwifi-hw$ ./prepare_adi_board_ip.sh $XILINX_DIR zc706_fmcs2
$XILINX_DIR is found!
./adi-hdl/projects/fmcomms2/zc706/
++ source /home/mars/Xilinx/Vivado//Vivado/2018.3/settings64.sh
+++ source /home/mars/Xilinx/Vivado/DocNav/.settings64-DocNav.sh
++++ '[' -n /home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /home/mars/Xilinx/Vivado/Vivado/2018.3/.settings64-Vivado.sh
++++ export XILINX_VIVADO=/home/mars/Xilinx/Vivado/Vivado/2018.3
++++ XILINX_VIVADO=/home/mars/Xilinx/Vivado/Vivado/2018.3
++++ '[' -n /home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/home/mars/Xilinx/Vivado/Vivado/2018.3/bin:/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/home/mars/Xilinx/Vivado/Vivado/2018.3/bin:/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /home/mars/Xilinx/Vivado/SDK/2018.3/.settings64-SDK_Core_Tools.sh
++++ '[' -n /home/mars/Xilinx/Vivado/Vivado/2018.3/bin:/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/home/mars/Xilinx/Vivado/SDK/2018.3/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/microblaze/lin/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/arm/lin/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/home/mars/Xilinx/Vivado/Vivado/2018.3/bin:/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/home/mars/Xilinx/Vivado/SDK/2018.3/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/microblaze/lin/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/arm/lin/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/mars/Xilinx/Vivado/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/home/mars/Xilinx/Vivado/Vivado/2018.3/bin:/home/mars/Xilinx/Vivado/DocNav:/home/mars/bin:/home/mars/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++ cd ./adi-hdl/projects/fmcomms2/zc706/
++ make
Building axi_ad9361 library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/axi_ad9361/axi_ad9361_ip.log] ... OK
Building axi_clkgen library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/axi_clkgen/axi_clkgen_ip.log] ... OK
Building util_cdc library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/util_cdc/util_cdc_ip.log] ... OK
Building util_axis_fifo library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/util_axis_fifo/util_axis_fifo_ip.log] ... OK
Building axi_dmac library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/axi_dmac/axi_dmac_ip.log] ... OK
[... cut ...]
Building util_wfifo library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/util_wfifo/util_wfifo_ip.log] ... OK
Building util_clkdiv library [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/library/xilinx/util_clkdiv/util_clkdiv_ip.log] ... OK
Building fmcomms2_zc706 project [/home/mars/open_wifi_5/openwifi-hw/adi-hdl/projects/fmcomms2/zc706/fmcomms2_zc706_vivado.log] ... OK
++ cd /home/mars/open_wifi_5/openwifi-hw

then I launch Vivado and select the cd /home/mars/open_wifi_5/openwifi-hw/boards/zc706_fmcs2/ and source path in TCL source ./openwifi.tcl
[image]

andrew712, 2021-06-22:
I found a problem, the error occurred due to the system language, apparently at some point it was switched and everything broke, thanks for the quick response

closed 2021-06-22 by andrew712