open-sdr/openwifi-hw #37: ERROR: [Common 17-54] The object 'project' does not have a property 'legacy_ip_repo_paths'.

lewiz-kiratey, 2021-07-19:
Hello,

I wanted to understand the design (IP and the project hierarchy) so I just wanted to load the design in vivado. I am getting the following error on 'source ./openwifi.tcl'.

```
# set_property -name "legacy_ip_repo_paths" -value "" -objects $obj
ERROR: [Common 17-54] The object 'project' does not have a property 'legacy_ip_repo_paths'.
```

Can this be because of the Vivado version? I am using the 2018.2 instead of 2018.3. Any help or source on just loading the design specific to ZCU102 would be helpful.

Thanks

JiaoXianjun, 2021-07-19:
Hi,

Sorry that we don't have your environment (vivado 2018.2). Would you please try 2018.3 and follow strictly our document?
If there is still issue with our versions, just let us know.

lewiz-kiratey, 2021-07-20:
Hello,

Thank you for your response, I was able to load the design with the Vivado 2018.3.

closed 2021-07-20 by lewiz-kiratey

lewiz-kiratey, 2021-07-23:
Hello,

Is there any documentation explaining the project (code structure and hierarchy)? After implementing the design, I am now able to see the Block Design and its interconnections. The top system design is easier to understand but, I wanted to learn more about the OPENWIFI_IP and XCU specific to ZCU102 board. Any help or guidance will be helpful.

Thanks

JiaoXianjun, 2021-07-23:
I think our readme already give very important info for understanding the project. 

You could ask concrete question instead of like “how to understand”.

lewiz-kiratey, 2021-08-16:
Hello,

I was finally able to build and implement the design with 'ADRV9361-Z7035 + ADRV1CRR-BOB/FMC' boards as the ZCU102 board that I have is 'xczu15eg' and your design seems compatible with the 'xczu9eg'.

I wanted to modify the hw-design a little and add some ILAs inside the 'openwifi_ip' package based on 'adc_clk' and 'm_axi_mm2s_aclk' clocks. After adding the ILAs and running the design again, I can't see ILAs coming up directly on Vivado. If I try to program the board through the Vivado Hardware Manager, I can see the ILAs but getting errors in running the 'openwifi_ap' '~/openwifi/fosdem-11ag.sh' on board. I believe the issue here is that I need some freerunning clocks for ILAs. Can I get some help on this? Also, can you please confirm some things as my understanding is that 'adc_clk' is 100MHz and 'm_axi_mm2s_aclk' is 200MHz for the 'openwifi_ip'?

Thanks so much.

JiaoXianjun, 2021-08-17:
Hello, 

The best way is to open new issue on github:
1. one issue for one question. don't mix questions
2. use our template

Thanks!