open-sdr/openwifi-hw #39: Add ILAs to the HW-Design

lewiz-kiratey, 2021-08-17:
Hello,

0. I am working with the boards 'ADRV9361-Z7035 + ADRV1CRR-BOB' and was successfully able to create 'openwifi' SSID, connect to it and load the local host page through phone. I am now trying to add some ILAs to the HW-Design inside the openwifi-ip based on the module inputs 'adc_clk' and 'm_axi_mm2s_aclk' clocks.

1. Our image is used directly or you build your own image?
Yes, your image is used directly.

2. What is your own modification?
Added some ILAs inside the openwifi-ip. If I just use the 'adc_clk' and 'm_axi_mm2s_aclk' clocks for the ILAs, after running the project on board and connecting the board to the system via JTAG, the ILAs does not load directly on the Vivado Hardware Manager. If I try to program the board from Vivado, then I can see the ILAs but I am getting errors in running the openwifi_ap '~/openwifi/fosdem-11ag.sh' on board.I believe the issue here is that I need some free-running clocks for ILAs. Can I get some help on this? Also, can you please confirm some things as my understanding is that 'adc_clk' is 100MHz and 'm_axi_mm2s_aclk' is 200MHz for the 'openwifi_ip'?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
UBUNTU 14.04
VIVADO 2018.3
Latest GitHub commit: 868aad30fe3cc1bee141fb21e2aa9a7cc51d8b12

4. Board/hardware type
'ADRV9361-Z7035 + ADRV1CRR-BOB'

5. WiFi channel number:
TX1A/RX1A

6. Steps to reproduce the issue, and the related error message, screenshot, etc
Add an ILA inside the openwifi-ip, with 3 inputs: 'dac_ready', 'dac_valid' & 'dac_data' and clock input 'adc_clk'. Mark all 3 inputs as 'debug'. Synthesize the design and follow the same steps for the 'ADRV9361-Z7035 + ADRV1CRR-BOB/FMC' project. Connect the board to the system with Vivado via JTAG cable, open Vivado Hardware Manager and 'auto-connect' to the board. Once the board is loaded, ideally, it should also load ILAs on the Vivado Hardware Manager. But, I can't see those so I think, I need free-running clocks coming from the board that is same as the 'adc_clk' frequency.

Thanks

JiaoXianjun, 2021-08-18:
"after running the project on board and connecting the board to the system via JTAG, the ILAs does not load directly on the Vivado Hardware Manager"

Did you load openwifi-hw/boards/adrv9361z7035/sdk/system_top.ltx to hardware manager? Without this ILA description, you can't see ila.

lewiz-kiratey, 2021-08-18:
I have tried loading the *.ltx files (system_top.ltx, debug_nets.ltx) in the hardware manager probe files section and refreshing the board. It still says there are no debug cores. I am attaching the screenshot of my vivado hardware manager window.

[image]

JiaoXianjun, 2021-08-18:
But in the screenshot, the "Probes file" are empty. Are you sure, you have add the correct file in that place? ("Probes file")

Did you follow this to get system_top.ltx?

In Vivado:
```
Open Block Design
Tools --> Report --> Report IP Status
Generate Bitstream
(Will take a while)
File --> Export --> Export Hardware... --> Include bitstream --> OK
File --> Launch SDK --> OK, then close SDK
```

In Linux:
```
cd openwifi-hw/boards
./sdk_update.sh board_name
```

To add ila, only mark the signal as debug is not enough. You need to add them manually after synthesis:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Then re-build the FPGA image, and run ./sdk_update.sh board_name. Pick the .ltx to the hardware manager "Probes file" tab.

lewiz-kiratey, 2021-08-21:
Hello,

Thank you very much for your time and help, also guiding me through these detailed steps. It was my reading error, for the last 2 builds that I have created, I missed out on the ' ./sdk_update.sh $BOARD_NAME ' step.

Although, now I can see all the ILAs, I am facing a different issue while running the '~/openwifi/fosdem-11ag.sh'. I was trying to debug that but in the meantime, I thought I should update you and see if you see something that's not right.

Issue: I boot up the board, run the '~/openwifi/fosdem-11ag.sh' and in the last few lines, I can also see:
```
Using interface sdr0 with hwaddr 66:55:44:33:22:25 and ssid "openwifi"
sdr0: interface state COUNTRY_UPDATE->ENABLED
sdr0: AP-ENABLED
```
But, when I try to look for 'openwifi' on phone or laptop, I can't find the network. I tried original BOOT.bin from the SD-CARD and it works (I am able to connect and load the default hosted page). If I update the BOOT.bin from the project that I have created (with ILAs), I can't find the 'openwifi' on any device wifi networks.

I am also attaching the putty capture using the updated design (with ILAs).

Thanks

[putty_openwifi.txt](https://github.com/open-sdr/openwifi-hw/files/7024485/putty_openwifi.txt)

JiaoXianjun, 2021-08-21:
Please follow this issue https://github.com/open-sdr/openwifi/issues/90

to debug. Check /proc/interrupts, etc. 

Also try not to insert ila, just re-generate our fpga to try.

lewiz-kiratey, 2021-08-23:
Thank you for your response @JiaoXianjun . I checked the 'cat /proc/interrupts' for the hw-designs (modified: with ILAs and original: without ILAs). I can see the 'sdr,tx_intrpt' incrementing for the original design and not incrementing for the modified design. Would you be able to guess what the issue might be? The only change between both sd-cards here is the BOOT.bin file in the 'BOOT' partition.
I am attaching the putty capture of the 'cat /proc/interrupts' as well as 'dmesg' output for the modified hw-design.

[sdr_tx_intr.txt](https://github.com/open-sdr/openwifi-hw/files/7033382/sdr_tx_intr.txt)

Thanks

JiaoXianjun, 2021-08-23:
Have you tried to generate BOOT.BIN withouting adding ila and without any change to the fpga code? Does this BOOT.BIN work?

(Don’t use our original BOOT.BIN. Generate yourself)

lewiz-kiratey, 2021-08-23:
Ah okay, thanks so much for your quick response. I get what you meant now and I tried that.
First, I tried the original image from you. Then ran the synthesis on the hw-design (without modifications). Generated bitstream and './sdk_update.sh board_name' step. After the board is ON, followed the Update FPGA steps: https://github.com/open-sdr/openwifi#update-fpga
With new BOOT.bin, the project still works fine. I am able to connect to the 'openwifi' network and load the local hostpage.
I am also attaching the putty capture for the new hw-design (without ILAs).

[sdr_tx_intr_orig.txt](https://github.com/open-sdr/openwifi-hw/files/7034140/sdr_tx_intr_orig.txt) : holds the capture on your original design.

[sdr_tx_intr_after_boot.txt](https://github.com/open-sdr/openwifi-hw/files/7034147/sdr_tx_intr_after_boot.txt) : capture after updating the FPGA.

Hope this helps.

Thanks again.

JiaoXianjun, 2021-08-23:
Now add ila to the working FPGA project you just created, try again.