open-sdr/openwifi-hw #44: Advise on connecting to the board.

lewiz-kiratey, 2021-09-10:
0. Hello, I have 2 separate sets of 'ADRV9361-Z7035 + ADRV1CRR-BOB' boards that has working openwifi. I wanted to use one of the boards as it is with running openwifi (host board) and the other board as client (connecting to the 'openwifi' network by the host board). Can you please give me some info on how this can be achieved?

1. Our image is used directly or you build your own image?
yes

2. What is your own modification?
none so far

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
UBUNTU 14.04
VIVADO 2018.3
Latest GitHub commit: 868aad30fe3cc1bee141fb21e2aa9a7cc51d8b12

4. Board/hardware type
two qty. 'ADRV9361-Z7035 + ADRV1CRR-BOB'

5. WiFi channel number
TX1A/RX1A

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
From the client board, I tried to look for the 'openwifi' network that has different MAC and IP ADDR for SDR0 port but are in the same subnet, still not able to connect if both the boards are ON and close to each other. At times, it will affect other WiFi connected devices (like laptop).

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better? 
Need the advice on how this connection can be achieved?

JiaoXianjun, 2021-09-12:
Please check our app notes: https://github.com/open-sdr/openwifi/tree/master/doc/app_notes

Especially this one: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/ap-client-two-sdr.md

lewiz-kiratey, 2021-09-13:
GREAT! Thank you so much @JiaoXianjun for your response. I was able to connect to the AP board from a client board as mentioned in steps in the second link you provided.

closed 2021-09-13 by lewiz-kiratey