open-sdr/openwifi-hw #45: Advise on verifying the direct TX/RX wireless path.

lewiz-kiratey, 2021-09-14:
Hello,

I wanted to verify the direct TX1A to RX1A wireless link. I am sending a stream of repeated data on the 'dac_data' bus (modified the 'tx_intf' module and added the ILAs on 'dac_data' and 'adc_data' buses). The 'dac_data' and the 'adc_data' buses are 64-bits. On the transmit side, the 'dac_data' only sends data on the lower 32-bits while the upper 32-bits are always zeros (only antenna-0 is used). On the receiving side, all 64-bits 'adc_data' come in randomly. I just wanted to see the data LOOPEDBACK from the TX1A to RX1A wirelessly. Can you please give me some advise on how to achieve this? As I am not able to find the transmitted data on the receive side.

1. Our image is used directly or you build your own image?
The FPGA was modified a little.

2. What is your own modification?
Modified the 'tx_intf' to send a stream of repetitive data pattern on 'dac_data' bus.

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
UBUNTU 14.04
VIVADO 2018.3
Latest GitHub commit: 868aad30fe3cc1bee141fb21e2aa9a7cc51d8b12

4. Board/hardware type
'ADRV9361-Z7035 + ADRV1CRR-BOB'

5. WiFi channel number
TX1A/RX1A

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
I tried this with your original image as well (after just adding the ILAs on 'dac_data' and 'adc_data' buses; no functional modifications). The entire 'openwifi' project was working. While the beacon pattern was continuously sent on the transmit side 'dac_data', I was not able to capture that pattern on the receive side 'adc_data' bus (tried to find one data at a time).

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2021-09-15:
When you capture adc_data via ila, what is the capture length and trigger condition?

lewiz-kiratey, 2021-09-15:
Hello @JiaoXianjun 

Thanks for the response.

The capture length for the ILAs are 8192. For the trigger condition, I captured the 'dac_data' going out for several different runs and at different time intervals. I saw a stream of data going out that starts with:

`
dac_data (HEX)
0000_0000_FF68_FF68
0000_0000_FDE0_0220
0000_0000_0410_0250
0000_0000_EC68_0D60
0000_0000_FCC8_07C0
0000_0000_D7AC_FDD0
0000_0000_F9C0_F5A8
0000_0000_046C_E974
0000_0000_07F4_F534
0000_0000_2638_0018
...
`

Although the periodic beacon pattern is as big as 5283 samples, the first 1600 data samples seemed the same for different captures of the 'dac_data'. So I am just putting this data as a trigger on the 'adc_data == 64'hXXXX_XXXX_FF68_FF68' and so on...
Although, I was able to find first 2 data, it was never consistent. Sometimes it would find immediately, sometimes it would take a while (Note: the periodic transmit send pattern is immediate). I can't find the 3rd data (0000_0000_0410_0250) and the data following it. To avoid any human error, I also tried to integrate a small module in the working openwifi design that can capture first 16 data samples on the 'dac_data' bus. But the found_count doesn't go beyond 3 samples.

Hope I am able to answer your questions correctly.

Thanks

JiaoXianjun, 2021-09-17:
If you are looping back via the antenna/air, the adc data won't match exactly to the dac data, because of the non-ideal, distortion of the RF and the air/channel.

I suggest you focus on the amplitude of adc data. Set adc data to signed decimal format in the ila window of vivado, and set trigger condition like > some_value. You can do free running trigger to see the usual/noise-floor value of adc data, then set a reasonable threshold value for your trigger on adc data to capture the self-tx-packet over the air.

lewiz-kiratey, 2021-09-17:
Yes, that's what I suspected afterwards. Thanks for letting me know. I can see a lot of processing being done in the rx_intf and opeofdm_rx modules. I tried to change the level where I was comparing the TX and RX data (instead of adc_data and dac_data). This time I tried to compare the I/Q data on both sides but I still believe that this is not the fully recovered data. So can you please confirm if the 802.11 pkt data at the output of openofdm_rx is the fully recovered version of the transmitted data? If not, where will I be able to see the recovered data?

I appreciate your help, thanks!

JiaoXianjun, 2021-09-17:
to let ofdm receiver decode the self-transmitted signal, you need to disable the self-muting in xpu:

./sdrctl dev sdr0 set reg xpu 1 1

You can just follow this app note to make sure the self-loop work: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi_fuzzer.md

lewiz-kiratey, 2021-09-24:
Hello @JiaoXianjun ,

Thank you very much for guiding me to the app note. I followed the steps and the config worked for me. I have a question on the type of modulation that we are doing on the outgoing data for this configuration: https://github.com/open-sdr/openwifi-hw/issues/45#issue-996400924

I was digging through some source codes and by modulation I mean inside the blocks: 'openofdm (dot11_tx.v) -> modulation (modulation.v)'. 'N_BPSC' value that determines whether the modulation is BPSK, QPSK, 16-QAM or 64-QAM?

Thanks

JiaoXianjun, 2021-09-30:
Linux decides the modulation when it calls: https://github.com/open-sdr/openwifi/blob/f2860429262ea644146d68e487279aacbf0d35de/driver/sdr.c#L655

The modulation is fetched by: rate_hw_value = ieee80211_get_tx_rate(dev, info)->hw_value;

You can override above to set the modulation you want at this driver level.

To understand the N_BPSC (BPSK, QPSK, 16-QAM or 64-QAM) of openofdm_tx in FPGA, the best way is to read the related chapter in the standard.

lewiz-kiratey, 2021-10-08:
Thank you very much @JiaoXianjun for your help.

closed 2021-10-08 by lewiz-kiratey