open-sdr/openwifi-hw #53: Problem with Vivado 2021.1 webpack, xc7z045ffg900-2 not supported

catkira, 2022-04-19:
I am already trying to get it running with a new Vivado version (2021.1 on Ubuntu 18).

However, source ./openofdm_rx.tcl from the script package_ip_openofdm_rx.tcl gives this error:
`ERROR: [Board 49-71] The board_part definition was not found for xilinx.com:zc706:part0:1.2. The project's board_part property was not set, but the project's part property was set to xc7z045ffg900-2. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.`

The problem is the following line:
create_project ${project_name} ./${project_name} -part xc7z045ffg900-2

I think it might be because this part is not available in Vivado webpack. Do You think we should make this "xc7z045ffg900-2" so that the correct FPGA for each board is selected? I am using antsdr, so a xc7z020 should be used.

JiaoXianjun, 2022-04-22:
Sorry that I don't have webpack version of 2021.1. Just tried a normal vivado 2019.1, and there is not error when I source the openofdm_rx.tcl in vivado.

Could you please confirm two things?
1. When you source ./openofdm_rx.tcl in your Vivado (not via package_ip_openofdm_rx.tcl), the error is still there?
2. When you change the part to -part xc7z020clg484-1, the error is gone?

catkira, 2022-04-23:
Sorry for the delay, I was travelling in UK...
1. yes
2. no, which is strange....
This is what I get in the TCL console:
`# create_project ${project_name} ./${project_name} -part xc7z020clg484-1
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2021.1/data/ip'.
create_project: Time (s): cpu = 00:00:17 ; elapsed = 00:00:07 . Memory (MB): peak = 7562.484 ; gain = 40.039 ; free physical = 29643 ; free virtual = 39068
# set proj_dir [get_property directory [current_project]]
# set obj [current_project]
# set_property -name "board_connections" -value "" -objects $obj
# set_property -name "board_part" -value "xilinx.com:zc706:part0:1.2" -objects $obj
ERROR: [Board 49-71] The board_part definition was not found for xilinx.com:zc706:part0:1.2. The project's board_part property was not set, but the project's part property was set to xc7z020clg484-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
`

catkira, 2022-04-23:
openofdm_rx.tcl runs fine if I out-comment this line
`
set_property -name "board_part" -value "xilinx.com:zc706:part0:1.2" -objects $obj
`

catkira, 2022-04-23:
But the change to xc7z020clg484-1 is needed anyway, because with the xc7z045ffg900-2 I get this synthesis error
```
[Common 17-345] A valid license was not found for feature 'Synthesis' and/or device 'xc7z045'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ". If you are using a license server, verify that the license server is up and running a version of the xilinx daemon that is compatible with the version of Xilinx software that you are using. Please note that Vivado 2017.3 and later requires upgrading your license server tools to the Flex 11.14.1 tools. Please confirm with your license admin that the correct version of the license server tools are installed.
```

JiaoXianjun, 2022-04-26:
Yes. For FPGA larger than 7020, Xilinx asks for a license. For FPGA <= 7020, Xilinx offer the FPGA tool for free.

JiaoXianjun, 2022-04-26:
> openofdm_rx.tcl runs fine if I out-comment this line `set_property -name "board_part" -value "xilinx.com:zc706:part0:1.2" -objects $obj`

Good to know this works. We will change accordingly. But still, we never try Vivado newer than 2018.3, so there might be other issues we never see. To be sure it works, 2018.3 is suggested.

catkira, 2022-04-26:
I think sticking to old Vivado 2018.3 is not a long term solution. I already know that there are other minor problems with the new Vivado, some of the Xilinx IPs have to be updated to a newer version. 
I can make a branch where I do all the changes needed for the latest Vivado and do a PR? Or do You want to create a branch?

JiaoXianjun, 2022-04-26:
We won't stick to 2018.3 for ever. The reason is that ADI HDL reference design (and ADI Linux kernel/devicetree/file-system variant) is based on specific version of Vivado explicitly, and all these infrastructures (HDL and Linux kernel/etc) are our starting point, so we'd better always aligne with ADI.

As you can see in this page: https://wiki.analog.com/resources/tools-software/linux-software/embedded_arm_images . 
Current openwifi is based on ( click Older releases (Click to expand) ):
22 June 2020 release (2019_R1)

Then check here: https://github.com/analogdevicesinc/hdl/releases/tag/2019_r1 , the HDL needs Vivado 2018.3.

Our next target is
28 July 2021 (2019_R2 Released)

Then check here: https://github.com/analogdevicesinc/hdl/releases/tag/2019_r2 , the Vivado needs to be 2019.1. So I am not sure Vivado 2021.1 is a good idea ...

Another reminding is that: upgrading to 2019_R2 is not just about HDL/Vivado, for openwifi there is also a big step of the Linux distribution version (from Linaro to raspbian Kuiper) and kernel and the devicetree and etc. We tried 2019_R2 (HDL and Linux) before on zcu102 as an initial experimental step, and it was basically successful (with some minor non-ideal things). But then we focus on optimizing openwifi IP itself, so that step is not continued so far.

It would be good that you can try the full upgraded solution (HDL and Linux of 2019_R2, preferably with Vivado 2019.1) on your board and let us know the result and necessary adjustments. Then we can take a look how to proceed.

Thanks!

catkira, 2022-04-26:
I expect Vivado 2022.1 to come out in the next couple of weeks. I will try to get it running then. 
Having so many different Vivado installations on a hard drive is a bit annoying, because each one is using about 100 GB and and also some language features (system verilog) are improved/added in newer versions.

catkira, 2022-04-28:
1. I need the latest vivado versions for other projects. Having to install several versions of vivado (and some of them on windows and linux) quickly fills up a hard drive
2. I like to use new language features that are not supported in older vivado versions
3. New versions of Vivado have more bug fixes (and sometimes also some new bugs)

JiaoXianjun, 2022-04-29:
Good.

For openwifi, we aim to offer stable and good WiFi experience, it is better for us to stick to ADI version. 

But if you have any suggestions for making the openwifi more compatible to newer version, just let us know and let’s see whether it is easy to be integrated. 

Thanks.

closed 2022-04-29 by JiaoXianjun

JiaoXianjun, 2023-01-28:
Here is the official release to support ADI/Xilinx new version:
https://github.com/open-sdr/openwifi/releases/tag/v1.4.0