open-sdr/openwifi-hw #54: Generate Bitstream error:module 'module 'system' not found [system_wrapper.v:269]

brealin, 2022-04-23:
Compiled openwifi-hw, pulled the latest github code, and followed the steps strictly. In the last step of "Generate Bitstream", the system module could not be found. See the attached file.
[image]

brealin, 2022-04-23:
bl@ubuntu:~/open-wifi$ rm -rf openwifi-hw/
bl@ubuntu:~/open-wifi$ ls
openwifi
bl@ubuntu:~/open-wifi$ git clone https://github.com/open-sdr/openwifi-hw.git
正克隆到 'openwifi-hw'...
remote: Enumerating objects: 4679, done.
remote: Counting objects: 100% (881/881), done.
remote: Compressing objects: 100% (289/289), done.
remote: Total 4679 (delta 600), reused 789 (delta 590), pack-reused 3798
接收对象中: 100% (4679/4679), 456.25 MiB | 298.00 KiB/s, 完成.
处理 delta 中: 100% (2833/2833), 完成.
bl@ubuntu:~/open-wifi$ cd openwifi-hw/
bl@ubuntu:~/open-wifi/openwifi-hw$ ./prepare_adi_lib.sh $XILINX_DIR
$XILINX_DIR is found!
++ git submodule init adi-hdl
子模组 'adi-hdl'（https://github.com/analogdevicesinc/hdl.git）未对路径 'adi-hdl' 注册
++ git submodule update adi-hdl
正克隆到 '/home/bl/open-wifi/openwifi-hw/adi-hdl'...
子模组路径 'adi-hdl'：检出 'f61d9707eb0a62533efd6facab59ab2444da94c9'
++ cd ./adi-hdl/library
++ git reset --hard f61d9707eb0a62533efd6facab59ab2444da94c9
HEAD 现在位于 f61d9707e ad_fmclidar1_ebz: Update the IO constraints to revB
++ cd /home/bl/open-wifi/openwifi-hw
bl@ubuntu:~/open-wifi/openwifi-hw$ ./prepare_adi_board_ip.sh $XILINX_DIR $BOARD_NAME
$XILINX_DIR is found!
./adi-hdl/projects/adrv9361z7035/ccbob_lvds/
++ source /tools/Xilinx/Vivado/2018.3/settings64.sh
+++ source /tools/Xilinx/DocNav/.settings64-DocNav.sh
++++ '[' -n /tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vivado/2018.3/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/tools/Xilinx/DocNav:/tools/Xilinx/SDK/2018.3/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/tools/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/ [... cut ...]
+ cd /home/bl/open-wifi/openwifi-hw
bl@ubuntu:~/open-wifi/openwifi-hw$ cd ..
bl@ubuntu:~/open-wifi$ cd openwifi-hw/boards/$BOARD_NAME/
bl@ubuntu:~/open-wifi/openwifi-hw/boards/adrv9361z7035$ source $XILINX_DIR/Vivado/2018.3/settings64.sh
bl@ubuntu:~/open-wifi/openwifi-hw/boards/adrv9361z7035$ vivado

****** Vivado v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

JiaoXianjun, 2022-04-23:
What have you done in Vivado?

brealin, 2022-04-23:
Follow the instructions to do the following:
1.source ./ip_repo_gen.tcl
(Generating ip_repo from ip design. Will take a while.)
2.source ./openwifi.tcl
3.Click "Generate Bitstream" in the Vivado GUI.

No changes have been made to the code. Just want to compile successfully once

JiaoXianjun, 2022-04-23:
Please try a fully English Linux system. We never try Chinese character Linux system

brealin, 2022-04-23:
Issue resolved, missing Xilinx Viterbi Decoder license issue mentioned in documentation

closed 2022-04-23 by brealin

chenmo234, 2024-05-29:
I am using E310 to deploy openwifi. Can I ask you some questions? Here is my contact information: c2385405729.

> 问题已解决，文档中提到的缺少 Xilinx Viterbi 解码器许可证问题