open-sdr/openwifi-hw #55: tx lo divider control register

xiaogang05, 2022-04-24:
in ad9361 manual the register of tx
 Lo control is 0x051,but in spi.v the register data is
"define SPI_HIGH 24'hC22001", reverse of spi data is 0x800443, the address is 0x004,why not 0x051？

HavingaThijs, 2022-04-26:
Another way to limit Tx leakage is to switch to RF port B during receiving, which is grounded on some boards. This has a slightly faster reaction time (since powering the Tx LO divider on takes about 160ns), but cannot totally suppress self-interference. 
The xpu_*.tcl sets the SPI data to control the RF port, which is address 0x004. In [ip_repo_gen.tcl](https://github.com/open-sdr/openwifi-hw/blob/master/boards/zcu102_fmcs2/ip_repo_gen.tcl), by default it is set to control the Tx LO divider (24'h088A01), which indeed is register 0x051. Hope this clarifies it.

xiaogang05, 2022-04-26:
Get it,thanks.

closed 2022-04-26 by xiaogang05