open-sdr/openwifi-hw #56: source ./ip_repo_gen.tcl will report error

cjhonlyone, 2022-05-14:
When I clone master branch directly, and following the README to use ip_repo_gen.tcl to generate project will report error. 

But if I clone master branch user --recursive parameter, the project will be built successfully.

I think the matter is the variable $argv must clear before source openofdm_rx.tcl.
./get_ip_openofdm_rx.sh link to branch dot11zynq made a mistake.

This is my vivado log.

`source ip_repo_gen.tcl
# exec rm -rf ip_repo
# exec mkdir ip_repo
# exec cp ../../ip/board_def.v ./ip_repo/ -f
# set  fd  [open  "./ip_repo/openwifi_hw_git_rev.v"  w]
# set HASHCODE [exec ../../get_git_rev.sh]
# puts $fd "`define OPENWIFI_HW_GIT_REV (32'h$HASHCODE)"
# close $fd
# set has_side_ch 1 
# set  fd  [open  "./ip_repo/has_side_ch_flag.v"  w]
# if {$has_side_ch > 0} {
#   puts $fd "`define HAS_SIDE_CH 1"
# } else {
#   puts $fd "`define NO_SIDE_CH 1"
# }
# close $fd
# set small_fpga 1
# set  fd  [open  "./ip_repo/fpga_scale.v"  w]
# if {$small_fpga == 1} {
#   puts $fd "`define SIDE_CH_LESS_BRAM 1"
# }
# close $fd
# set NUM_CLK_PER_US 100
# set  fd  [open  "./ip_repo/clock_speed.v"  w]
# puts $fd "`define NUM_CLK_PER_US $NUM_CLK_PER_US"
# if {$small_fpga == 1} {
#   puts $fd "`define SMALL_FPGA 1"
# }
# close $fd
# set grounded_rf_port 0
# set  fd  [open  "./ip_repo/spi_command.v"  w]
# if {$grounded_rf_port == 1} {
#   puts $fd "`define SPI_HIGH 24'hC22001"
#   puts $fd "`define SPI_LOW 24'hC02001"
# } else {
#   puts $fd "`define SPI_HIGH 24'h088A01"
#   puts $fd "`define SPI_LOW 24'h008A01"
# }
# close $fd
# set ultra_scale_flag 0
# set part_string xc7z020clg484-1
# set argc 3
# set ip_name openofdm_rx
# exec rm -rf project_1
# set current_dir [pwd]
# set argv [list $ultra_scale_flag $current_dir/../../ip/$ip_name $current_dir/ip_repo/$ip_name]
# source ../package_ip_openofdm_rx.tcl
## set ultra_scale_flag [lindex $argv 0]
## set src_dir [lindex $argv 1]
## set ip_dir [lindex $argv 2]
## exec rm -rf project_1
## exec rm -rf $ip_dir
## set current_dir [pwd]
## cd $src_dir/
## if {$ultra_scale_flag > 0} {
##     exec rm -rf openofdm_rx_ultra_scale
##     source ./openofdm_rx_ultra_scale.tcl
## } else {
##     exec rm -rf openofdm_rx
##     source ./openofdm_rx.tcl
## }
### set ARGUMENT1 [lindex $argv 0]
### set ARGUMENT2 [lindex $argv 1]
### set ARGUMENT3 [lindex $argv 2]
### set ARGUMENT4 [lindex $argv 3]
### set ARGUMENT5 [lindex $argv 4]
### set ARGUMENT6 [lindex $argv 5]
### set ARGUMENT7 [lindex $argv 6]
### if {$ARGUMENT1 eq ""} {
###   set BOARD_NAME zed_fmcs2
### } else {
###   set BOARD_NAME $ARGUMENT1
### }
### if {$ARGUMENT2 eq ""} {
###   set NUM_CLK_PER_US 100
### } else {
###   set NUM_CLK_PER_US $ARGUMENT2
### }
### source ./parse_board_name.tcl
#### if {$BOARD_NAME=="zed_fmcs2"} {
####    set ultra_scale_flag 0
####    set part_string xc7z020clg484-1
####    set fpga_size_flag 0
#### } elseif {$BOARD_NAME=="zcu102_fmcs2"} {
####    set ultra_scale_flag 1
####    set part_string xczu9eg-ffvb1156-2-e
####    set fpga_size_flag 1
#### } elseif {$BOARD_NAME=="zc706_fmcs2"} {
####    set ultra_scale_flag 0
####    set part_string xc7z045ffg900-2
####    set fpga_size_flag 1
#### } elseif {$BOARD_NAME=="zc702_fmcs2"} {
####    set ultra_scale_flag 0
####    set part_string xc7z020clg484-1
####    set fpga_size_flag 0
#### } elseif {$BOARD_NAME=="antsdr"} {
####    set ultra_scale_flag 0
####    set part_string xc7z020clg400-1
####    set fpga_size_flag 0
#### } elseif {$BOARD_NAME=="adrv9361z7035"} {
####    set ultra_scale_flag 0
####    set part_string xc7z035ifbg676-2L
####    set fpga_size_flag 1
#### } elseif {$BOARD_NAME=="adrv9364z7020"} {
####    set ultra_scale_flag 0
####    set part_string xc7z020clg400-1
####    set fpga_size_flag 0
#### } else {
####    set ultra_scale_flag []
####    set part_string []
####    set fpga_size_flag []
####    puts "$BOARD_NAME is not valid!"
#### }
0 is not valid!
### set MODULE_NAME OPENOFDM_RX
### set  fd  [open  "./verilog/openofdm_rx_pre_def.v"  w]
### if {$NUM_CLK_PER_US == 100} {
###   puts $fd "`define CLK_SPEED_100M"
### } elseif {$NUM_CLK_PER_US == 200} {
###   puts $fd "`define CLK_SPEED_200M"
### } elseif {$NUM_CLK_PER_US == 240} {
###   puts $fd "`define CLK_SPEED_240M"
### } elseif {$NUM_CLK_PER_US == 400} {
###   puts $fd "`define CLK_SPEED_400M"
### } else {
###   throw {NUM_CLK_PER_US MUST BE 100/200/240/400!}
### }
invalid command name "throw"
    while executing
"throw {NUM_CLK_PER_US MUST BE 100/200/240/400!}"
    invoked from within
"if {$NUM_CLK_PER_US == 100} {
  puts $fd "`define CLK_SPEED_100M"
} elseif {$NUM_CLK_PER_US == 200} {
  puts $fd "`define CLK_SPEED_200M"
} elseif {$N..."
    (file "./openofdm_rx.tcl" line 51)
Vivado% exit
`

JiaoXianjun, 2022-05-14:
Hello,

Thanks for the feedback. We just improved the ip/fpga project flow to support conditional compile and dynamic fpga/driver reloading (no need to reboot): https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/drv_fpga_dynamic_loading.md 

Please follow the updated README of openwifi-hw: https://github.com/open-sdr/openwifi-hw/blob/master/README.md

Should you still have problem, just let us know!

cjhonlyone, 2022-05-15:
The problem has been solved.
Thanks for replying and upgrading so quickly!!!

closed 2022-05-15 by cjhonlyone