open-sdr/openwifi-hw #58: Support for PlutoSDR

catkira, 2022-05-15:
I think it should be possible to get openwifi running on plutosdr, if the sidechannel could be disabled with a parameter. I already found some places where ressource usage could be reduced, the iq-balance and dc offset correction in the AD936x hdl can be deactivated by parameters, they are not used anyway, because AD936x implements these features in hardware. There is also an additional 128 tap fir filter in the ad936x hdl code which is not used, that could also be eliminated.

Having some support for PlutoSDR would be very cool, because it is the cheapest AD936x hardware by far (250 USD).

JiaoXianjun, 2022-05-15:
Indeed.
Looking forward to your good news!

catkira, 2022-05-15:
My idea was a bit premature, openofdm_rx consumes too many luts (15282) and the 7010 only has 17600 luts.

JiaoXianjun, 2022-05-15:
lut, dsp, bram, can be transformed between them if the resource is not well balanced. 

But, I can imagine, even it fits, it will be marginal.

catkira, 2022-05-16:
Your paper from 2017 is interesting. Back then You used the S_AXI_HP port of the PS to benchmark latency.

In this https://j-marjanovic.io/exploring-the-ps-pl-axi-interfaces-on-zynq-ultrascale-mpsoc.html measurement it is shown, that using the S_AXI_ACP has about 7x less latency. I havent used the ACP interface before, but the measurements shown in the link above make it very interesting.

JiaoXianjun, 2022-05-16:
If I remember correctly we also use ACP in some bd now. And we did some profiling back to then.  Latency is not the only thing, though it is important.