open-sdr/openwifi-hw #59: Bitstream Generation Problem

chubohao, 2022-06-13:
[image]

Hello,

I did all step successfully, but when I generated the bitstream, I have these problem? Do you konw what should I do?

Thanks in advance

Bohao

JiaoXianjun, 2022-06-13:
If you did all steps correctly, then you should not have any issue.

To answer your question, you have to paste here what you have done exactly in each step starting from "git clone": what is your command, what is the output, etc.

Only having that, we can help you.

chubohao, 2022-06-13:
> 
**### Here is all message of tcl console**
start_gui
source ./ip_repo_gen.tcl
# exec rm -rf ip_repo
# exec mkdir ip_repo
# exec cp ../../ip/board_def.v ./ip_repo/ -f
# set  fd  [open  "./ip_repo/openwifi_hw_git_rev.v"  w]
# set HASHCODE [exec ../../get_git_rev.sh]
# puts $fd "`define OPENWIFI_HW_GIT_REV (32'h$HASHCODE)"
# close $fd
# set has_side_ch 1 
# set  fd  [open  "./ip_repo/has_side_ch_flag.v"  w]
# if {$has_side_ch > 0} {
#   puts $fd "`define HAS_SIDE_CH 1"
# } else {
#   puts $fd "`define NO_SIDE_CH 1"
# }
# close $fd
# set small_fpga 1
# set  fd  [open  "./ip_repo/fpga_scale.v"  w]
# if {$small_fpga == 1} {
#   puts $fd "`define SIDE_CH_LESS_BRAM 1"
# }
# close $fd
# set NUM_CLK_PER_US 100
# set  fd  [open  "./ip_repo/clock_speed.v"  w]
# puts $fd "`define NUM_CLK_PER_US $NUM_CLK_PER_US"
# if {$small_fpga == 1} {
#   puts $fd "`define SMALL_FPGA 1"
# }
# close $fd
# set grounded_rf_port 0
# set  fd  [open  "./ip_repo/spi_command.v"  w]
# if {$grounded_rf_port == 1} {
#   puts $fd "`define SPI_HIGH 24'hC22001"
#   puts $fd "`define SPI_LOW 24'hC02001"
# } else {
#   puts $fd "`define SPI_HIGH 24'h088A01"
#   puts $fd "`define SPI_LOW 24'h008A01"
# }
# close $fd
# set BOARD_NAME [lindex [split [exec pwd] /] end]
# puts $BOARD_NAME
antsdr
# set ultra_scale_flag 0
# set part_string xc7z020clg400-1
# set argc 4
# set ip_name openofdm_rx
# if {[file exists ./ip_config/$ip_name\_pre_def.v]==0} {file mkdir ip_config; exec echo "" > ./ip_config/$ip_name\_pre_def.v}
# exec rm -rf project_1
# set current_dir [pwd]
# set argv [list $ultra_scale_flag $current_dir/../../ip/$ip_name $current_dir/ip_repo/$ip_name $BOARD_NAME]
# source ../package_ip_openofdm_rx.tcl
## set ultra_scale_flag [lindex $argv 0]
## set src_dir [lindex $argv 1]
## set ip_dir [lindex $argv 2]
## set BOARD_NAME [lindex $argv 3]
## exec rm -rf project_1
## exec rm -rf $ip_dir
## set current_dir [pwd]
## cd $src_dir/
## set argc 1
## set argv [list $BOARD_NAME]
## source ./openofdm_rx.tcl
### set ARGUMENT1 [lindex $argv 0]
### set ARGUMENT2 [lindex $argv 1]
### set ARGUMENT3 [lindex $argv 2]
### set ARGUMENT4 [lindex $argv 3]
### set ARGUMENT5 [lindex $argv 4]
[... 2426 lines cut ...]
# startgroup
# set_property -dict [list CONFIG.JITTER_SEL {No_Jitter} CONFIG.CLKOUT1_REQUESTED_OUT_FREQ {100} CONFIG.MMCM_DIVCLK_DIVIDE {1} CONFIG.MMCM_BANDWIDTH {OPTIMIZED} CONFIG.MMCM_CLKFBOUT_MULT_F {25.000} CONFIG.MMCM_CLKOUT0_DIVIDE_F {10.000} CONFIG.CLKOUT1_JITTER {180.876} CONFIG.CLKOUT1_PHASE_ERROR {191.950}] [get_bd_cells clk_wiz_0]
# endgroup
# report_ip_status -name ip_status 
# upgrade_ip [get_ips  {system_rx_intf_0_0 system_tx_intf_0_0 system_openofdm_tx_0_0 system_xpu_0_0 system_side_ch_0_0}] -log ip_upgrade.log
WARNING: [Coretcl 2-1042] No IP was identified for upgrade.
# export_ip_user_files -of_objects [get_ips {system_rx_intf_0_0 system_tx_intf_0_0 system_openofdm_tx_0_0 system_xpu_0_0 system_side_ch_0_0}] -no_script -sync -force -quiet
# report_ip_status -name ip_status 
# save_bd_design
Wrote  : </home/bohao/fpga/openwifi-hw/boards/antsdr/src/system.bd> 
Wrote  : </home/bohao/fpga/openwifi-hw/boards/antsdr/src/ui/bd_c954508f.ui> 
update_compile_order -fileset sources_1
launch_runs impl_1 -to_step write_bitstream -jobs 8
WARNING: [Runs 36-337] The following IPs are either missing output products or output products are not up-to-date for Synthesis target. Since these IPs are locked, no update to the output products can be done. 
Please select 'Report IP Status' from the 'Tools/Report' menu or run Tcl command 'report_ip_status' for more information.
/home/bohao/fpga/openwifi-hw/boards/antsdr/src/system.bd

WARNING: [Runs 36-337] The following IPs are either missing output products or output products are not up-to-date for Implementation target. Since these IPs are locked, no update to the output products can be done. 
Please select 'Report IP Status' from the 'Tools/Report' menu or run Tcl command 'report_ip_status' for more information.
/home/bohao/fpga/openwifi-hw/boards/antsdr/src/system.bd

[Mon Jun 13 08:02:00 2022] Launched synth_1...
Run output will be captured here: /home/bohao/fpga/openwifi-hw/boards/antsdr/openwifi_antsdr/openwifi_antsdr.runs/synth_1/runme.log
[Mon Jun 13 08:02:00 2022] Launched impl_1...
Run output will be captured here: /home/bohao/fpga/openwifi-hw/boards/antsdr/openwifi_antsdr/openwifi_antsdr.runs/impl_1/runme.log

chubohao, 2022-06-13:
> what about "Pre-conditions" we listed in the README?

Ohhh! 
'' Install the evaluation license of [Xilinx Viterbi Decoder] into Vivado. "

Thanks Dr.Jiao

closed 2022-06-13 by chubohao

chenmo234, 2024-05-29:
I am  using E310 to deploy openwifi. Can I ask you some questions? Here is my contact information: c2385405729.