open-sdr/openwifi-hw #60: Extracting samples after CFO correction

Aaron3219, 2022-06-28:
Hello,

I am currently trying to extract the samples after the CFO correction out of Verilog to the processor and to print them to a console or similar. I am using the Zedboard with AD-FMCOMMS3.
My first question is, if there is already an easy way to extract those samples? Maybe there is already an interface which I overlooked?

If there is not an interface, what would you consider the easiest way to accomplish my goal?
In Vivado I already created the outputs for everything and connected an AXI DMA:
[image]
You can see that I created `tdata_rotated`, `tvalid_rotated` and so on. However, I struggle to find an easy way to read the registers on the AXI DMA with C.

I also thought about connecting the newly created pins to the already existing AXI DMA and to change the `sdr.c` accordingly. But before I try this or more, I would like to hear your opinion about my questions.

JiaoXianjun, 2022-07-14:
The "rotate" instance inside sync_long.v does the CFO correction on IQ sample. You need to consider taking the output of that instance.

Aaron3219, 2022-07-14:
Thank you for your answer.
I tried adding a AXI DMA in the hardware design with the exact configuration. The DMA is in the device tree of linux and the IRQs are getting registered by the xlnx_dma driver. 

So far so good.

However, the interrupts by the new AXI DMA I added are not received by linux (I know for sure that I configured the interrupts in a right way in Linux and the concat block). Through debugging I determined that the configuration of the S_AXI_LITE connection is probably wrong.

So my follow up question is: Where do you set the values for the S_AXI_LITE connection for the DMAs?
They are coming out of the axi_interconnect1 block but where are you setting them in your code? I haven't been able to find it.

JiaoXianjun, 2022-07-14:
You can find what have done for the side_ch (fpga module and driver for side channel to extract information from FPGA).
You mean the interrupt number in Linux? I do not remember clearly. There might be a fixed offset. You need to check the Zynq ARM manual, and check the relationship of our existing design, such as tx_intf, rx_intf, side_ch. They all have AXI DMA and interrupt.

If you are talking about the base address of registers you add via S_AXI_LITE, you can check/change them in the Vivado block diagram design window. Again, you can learn/check our existing modules.