open-sdr/openwifi-hw #68: How to expand the data bit width of Side_Info from 64bit to 128bit?

KrisYuFi, 2022-09-11:
Hello, Dr. Jiao.
We are using your image.
**My modification**
    I saw that the top file for **side_ch** had a predefined data bit width and thought I could modify it directly. So, I tried modifying the **side_ch** IP Core of the FPGA , I want to use it to get more information at the same time, because I want to add some functionality, I tried to change the bit width of the data from side_ch to PS using DMA directly from 64bit to 128bit, and the width of the device tree to 128bit, but when using Linux script:`./side_ch_ctl g`.The **side_info_count** does not increase, which means no data send to PS.
[image]
[image]
[image]
[image]

**Our Board/hardware type:**
     Based on fcms2+zedboard.

**My debug efforts:**
      After a lot fruitless attempts, I looked at some of the signals using ILA and found that the AXI, which might be used to control the SIDE_CH cycle for data acquisition, might not be working properly.
[image]
The above is the online Debug diagram of my modified FPGA project.
[image]
The above is yours.
        The key is that the signal line slv_reg_wren, which should appear periodically.After the modification, it only becomes high the first time I execute `insmod side_ch.ko` and`./side_ch_ctrl g.`
**I guess**:Perhaps AXI is not working properly because DMA is passing 128 bits at a time but only 64bit is coming out?
Could you give me some suggestions about changing the data bit width?

JiaoXianjun, 2022-09-14:
First of all, check your ARM CPU spec of your board, to confirm it support 128bit DMA or not.

Instead of extend 64 to 128, you should consider change the RAM and state machine inside side_ch (and python script on computer) to put more 64bits into single DMA transaction (more 64bit words per transaction).

KrisYuFi, 2022-09-14:
> First of all, check your ARM CPU spec of your board, to confirm it support 128bit DMA or not.
>
> Instead of extend 64 to 128, you should consider change the RAM and state machine inside side_ch (and [...]

Thank you. I'll give it a try. I've definitely got more 64bit data.

closed 2022-11-01 by JiaoXianjun