open-sdr/openwifi-hw #86: openwifi compatible with vivado 2021 ML edition

sdrlab, 2023-01-18:
Hello ,
is there a way to make my openwifi HDL project using vivado version 2021 ML edition ?

JiaoXianjun, 2023-01-18:
It is done in the branch pre-release, which will be the candidate of our next release very soon.

You can already try it out in the pre-release branch now.

We will announce the test campaign soon, and we will very appreciate that if you can test that branch now. 

Thanks!

sdrlab, 2023-01-18:
thanks for the fast response.
if i want to build the project on widows how can i do it ? is there something i need to configure on cygwin in order for the make process complete ?

JiaoXianjun, 2023-01-18:
I never use windows, so I don’t know.

If you figure out how to do it, would you please share with the community?

sdrlab, 2023-01-19:
hello, i tried to start building the vivado project with the pre-release branch(it also happend in the master branch) and after using the ./prepare_adi_lib.sh $XILINX_DIR i got the following error : 

dhell@dror04 ~
$ cd /cygdrive/c

dhell@dror04 /cygdrive/c
$ cd openwifi-hw/

dhell@dror04 /cygdrive/c/openwifi-hw
$ ls
CONTRIBUTING.md  boards                 openwifi-logo.png
LICENSE          get_git_rev.sh         prepare_adi_board_ip.sh
README.md        get_ip_openofdm_rx.sh  prepare_adi_lib.sh
adi-hdl          gpio_led.md
bb-clk.jpg       ip

dhell@dror04 /cygdrive/c/openwifi-hw
$ git checkout pre-release
Already on 'pre-release'
Your branch is up to date with 'origin/pre-release'.

dhell@dror04 /cygdrive/c/openwifi-hw
$ cd C:\Xilinx

dhell@dror04 /cygdrive/c/Xilinx
$ cd ..

dhell@dror04 /cygdrive/c
$ cd openwifi-hw/

dhell@dror04 /cygdrive/c/openwifi-hw
$ export XILINX_DIR=C:\Xilinx

dhell@dror04 /cygdrive/c/openwifi-hw
$ ./prepare_adi_lib.sh $XILINX_DIR
$XILINX_DIR is not correct. Please check!

dhell@dror04 /cygdrive/c/openwifi-hw
$ export XILINX_DIR=C:/Xilinx

dhell@dror04 /cygdrive/c/openwifi-hw
$ ./prepare_adi_lib.sh $XILINX_DIR
$XILINX_DIR is found!
++ git submodule init adi-hdl
Submodule 'adi-hdl' (https://github.com/analogdevicesinc/hdl.git) registered for path 'adi-hdl'
++ git submodule update adi-hdl
Cloning into '/cygdrive/c/openwifi-hw/adi-hdl'...
Submodule path 'adi-hdl': checked out 'f61d9707eb0a62533efd6facab59ab2444da94c9'
++ cd ./adi-hdl/
++ git reset --hard
HEAD is now at f61d9707e ad_fmclidar1_ebz: Update the IO constraints to revB
++ git fetch
++ git checkout 2021_r1
Updating files: 100% (1474/1474), done.
Previous HEAD position was f61d9707e ad_fmclidar1_ebz: Update the IO constraints to revB
HEAD is now at d09fc9207 util_do_ram: Fix Rx path for interrupted transfers
++ git reset --hard 2021_r1
HEAD is now at d09fc9207 util_do_ram: Fix Rx path for interrupted transfers
++ source C:/Xilinx/Vivado/2021.1/settings64.sh
+++ source 'C:\Xilinx\Vitis\2021.1\.settings64-Vitis.sh'
++++ export 'XILINX_VITIS=C:\Xilinx\Vitis\2021.1'
++++ XILINX_VITIS='C:\Xilinx\Vitis\2021.1'
++++ export RDI_PLATFORM=win64
++++ RDI_PLATFORM=win64
++++ '[' -n '/usr/local/bin:/usr/bin:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/MATLAB/R2022b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2022b/bin:/cygdrive/c/Users/dhell/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/dhell/AppData/Local/Programs/Microsoft VS Code/bin:/cygdrive/c/Xilinx/Vivado/2018.3/bin:/cygdrive/c/Xilinx/Vivado_HLS/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/nt/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/arm/nt/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_be/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_le/bin:/cygdrive/c/Xilinx/SDk/2018.3/gnu/aarch32/nt/gcc-arm-none-eabi/bin' ']'
++++ export 'PATH=C:\Xilinx\Vitis\2021.1\bin:C:\Xilinx\Vitis\2021.1\gnu\microblaze\nt\bin:C:\Xilinx\Vitis\2021.1\gnu\arm\nt\bin:C:\Xilinx\Vitis\2021.1\gnu\microblaze\linux_toolchain\nt64_le\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch32\nt\gcc-arm-linux-gnueabi\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch32\nt\gcc-arm-none-eabi\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch64\nt\aarch64-linux\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch64\nt\aarch64-none\bin:C:\Xilinx\Vitis\2021.1\gnu\armr5\nt\gcc-arm-none-eabi\bin:C:\Xilinx\Vitis\2021.1\tps\win64\cmake-3.3.2\bin:C:\Xilinx\Vitis\2021.1\gnuwin\bin:/usr/local/bin:/usr/bin:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/MATLAB/R2022b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2022b/bin:/cygdrive/c/Users/dhell/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/dhell/AppData/Local/Programs/Microsoft VS Code/bin:/cygdrive/c/Xilinx/Vivado/2018.3/bin:/cygdrive/c/Xilinx/Vivado_HLS/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/nt/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/arm/nt/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_be/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_le/bin:/cygdrive/c/Xilinx/SDk/2018.3/gnu/aarch32/nt/gcc-arm-none-eabi/bin'
++++ PATH='C:\Xilinx\Vitis\2021.1\bin:C:\Xilinx\Vitis\2021.1\gnu\microblaze\nt\bin:C:\Xilinx\Vitis\2021.1\gnu\arm\nt\bin:C:\Xilinx\Vitis\2021.1\gnu\microblaze\linux_toolchain\nt64_le\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch32\nt\gcc-arm-linux-gnueabi\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch32\nt\gcc-arm-none-eabi\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch64\nt\aarch64-linux\bin:C:\Xilinx\Vitis\2021.1\gnu\aarch64\nt\aarch64-none\bin:C:\Xilinx\Vitis\2021.1\gnu\armr5\nt\gcc-arm-none-eabi\bin:C:\Xilinx\Vitis\2021.1\tps\win64\cmake-3.3.2\bin:C:\Xilinx\Vitis\2021.1\gnuwin\bin:/usr/local/bin:/usr/bin:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Program Files/MATLAB/R2022b/runtime/win64:/cygdrive/c/Program Files/MATLAB/R2022b/bin:/cygdrive/c/Use [... cut ...] bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_be/bin:/cygdrive/c/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/nt64_le/bin:/cygdrive/c/Xilinx/SDk/2018.3/gnu/aarch32/nt/gcc-arm-none-eabi/bin'
++ cd library/
++ make
lib was unexpected at this time.
make[1]: *** [../scripts/library.mk:79: xilinx_dep] Error 255
make: *** [Makefile:135: lib] Error 2
++ cd /cygdrive/c/openwifi-hw

dhell@dror04 /cygdrive/c/openwifi-hw
$

what can be the problem ? is it a problem in some way ?

JiaoXianjun, 2023-01-24:
> hello, i tried to start building the vivado project with the pre-release branch(it also happend in the master branch) and after using the ./prepare_adi_lib.sh $XILINX_DIR i got the following error [...]

I never use windows, so I don’t know.

If you figure out how to do it, would you please share with the community?

JiaoXianjun, 2023-01-28:
Here is the official release to support ADI/Xilinx new version:
https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

closed 2023-02-02 by JiaoXianjun