open-sdr/openwifi-hw #89: question

huaxueyong, 2023-03-27:
Hello Dr. Jiao! I am learning about your openwifi project. When I saw the csma module, I found that the backoff time directly began to decrease without waiting for DIFS. Is this in conflict with the violation?Thank you for taking time out of your busy schedule to check this message.

JiaoXianjun, 2023-03-28:
Can you give me the example code lines?

huaxueyong, 2023-03-28:
I looked at 80211-2016 Figure 10-15—Backoff procedure

JiaoXianjun, 2023-03-28:
Can you explain a bit more by an example of state transition chain?

huaxueyong, 2023-03-28:
time_run：begin
	if(channel_busy)
		state <= BACKOFF_SUSPEND ;
	else
		backoff_time <= backoff_time - 1 ;
end
time_stop:begin
	if(channel_busy)
		backoff_time <= backoff_time ;
	else begin
		if(wait_time == DIFS)begin
			state <= time_run ;
		else
			wait_time <= wait_time + 1 ;
	end
end
I think the system should have a process of waiting for DIFS instead of jumping to the next state directly.

huaxueyong, 2023-03-28:
[image]

huaxueyong, 2023-03-29:
Sorry, Dr. Jiao. Is there a problem with my answer？

JiaoXianjun, 2023-03-30:
I don't understand why you make the conclusion that we don't have DIFS. Can you give me an example by describing a state transition chain?

huaxueyong, 2023-04-01:
You only wait for one DIFS before setting the backoff time, but according to 802.11, backoff time still has to wait for one DIFS between stopping and running.

JiaoXianjun, 2023-04-03:
You are right. What you are saying actually is the after frame DIFS. We don't have that. That is one deviation from the standard.

To reduce FPGA logic resources occupancy, openwifi only has one CSMA/CA engine (serve all queues) and only start CSMA/CA engine if there is a packet waiting for transmission in a queue. For the DIFS in between stop and running (station B in your case), because there is no packet waiting for transmission, the CSMA/CA even doesn't run, so there is no DIFS in our implementation.

huaxueyong, 2023-04-03:
Thank you for your earnest answer to my question, but I think station B has packet to send, otherwise it would not set backoff time.

huaxueyong, 2023-04-03:
The stations in the figure represent different devices, not different queues under the same device.

huaxueyong, 2023-04-03:
For example, if two devices AB want to send data, they will both set backoff time(bt). Assuming that A preempts the channel first, then B_bt will stop and will not run until A's operation is completed and a DIFS is over.

huaxueyong, 2023-04-04:
I see.Thank you, Dr. Jiao. I believe openwifi will go further in your hands. openwifi forever

closed 2023-04-07 by JiaoXianjun