open-sdr/openwifi-hw #90: Hello,Dr.jiao. I have a question about the contention window of the csma code. I know that the traditional backoff window CWmin for CSMA/CA is 16, but the random number generation section in the csma_ca code starts from 2. How is this section defined? Or that I misunderstand the meaning of the source code?

liulangkk, 2023-04-01:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2023-04-03:
Can you elaborate in more detailed style? Such as which line of code in which kind of state transition in CSMA/CA part?

liulangkk, 2023-04-03:
https://github.com/open-sdr/openwifi-hw/blob/master/ip/xpu/src/csma_ca.v 
line239-279 is random number generator. In my view I know that it corresponds to the different stage of the backoff.
According to the cw_exp module, it can be seen that based on the retransmission trigger, the cw_exp+1 operation will be executed, which corresponds to the random number generation part in csma_ca. According to the CSMA/CA protocol (as described in most publicly available sources), the minimum contention window is randomly selected from within 16. My question is regarding the random number generation code in which, when cw_exp_used is set to 1, 2, or 3, the corresponding random number ranges are 2, 4, and 8, respectively. This may differ slightly from what I understand about the protocol, or this design may have a specific purpose, or perhaps I have misunderstood something.

HavingaThijs, 2023-04-06:
That part ensures that every possible value of the CW exponential can be used. While CWmin for the normal Distributed Coordination Function (DCF) is 15 (2^4-1), for Enhanced Distributed Channel Access (EDCA) it depends on the QoS type and may be lower than 15.
CWmin and CWmax are decided by Linux’ mac80211 or it can be overwritten by register 19 of the xpu (see https://github.com/open-sdr/openwifi/blob/master/doc/README.md#get-and-set-a-register-of-a-module).

liulangkk, 2023-04-11:
Thank you for your patient answer! This answer helps me a lot understand the meaning of the csma/ca code.

closed 2023-04-15 by liulangkk