open-sdr/openwifi-hw #96: antsdr e310v2 pull request

JiaoXianjun, 2023-05-08:
@black-pigeon 

I am trying to take e310v2 changes from your antsdr branch. Since in our side we have already moved fpga img files to openwifi-hw-img repository and we do not have sdk directory in openwifi-hw/boards/antsdr_e200 anymore, so would you please remove the sdk directory in openwifi-hw/boards/antsdr_e200 after this commit:
https://github.com/MicroPhase/openwifi-hw/commit/c66c0016f12b3090c57660115fcb8ec15187e7c3

Then I can check whether this will bring us a cleaner merge via the pull request that I already created.

black-pigeon, 2023-05-08:
OK, I will remove this directory.

JiaoXianjun, 2023-05-08:
Do you need to change this file of e200?
https://github.com/open-sdr/openwifi-hw/pull/95/files#diff-8c117a2a41539c1a2ce4c263d7448427b75d9185b9e4fc4407264267e42f0615

If not, could you reset it or check out it?

JiaoXianjun, 2023-05-08:
> Do you need to change this file of e200? https://github.com/open-sdr/openwifi-hw/pull/95/files#diff-8c117a2a41539c1a2ce4c263d7448427b75d9185b9e4fc4407264267e42f0615
> 
> If not, could you reset it or check out it?

Do you need to change this file of e200?
https://github.com/open-sdr/openwifi-hw/pull/95/files#diff-8c117a2a41539c1a2ce4c263d7448427b75d9185b9e4fc4407264267e42f0615

If not, could you reset it or check out it?

JiaoXianjun, 2023-05-08:
> OK, I will remove this directory.

Do you need to change this file of e200?
https://github.com/open-sdr/openwifi-hw/pull/95/files#diff-8c117a2a41539c1a2ce4c263d7448427b75d9185b9e4fc4407264267e42f0615

If not, could you reset it or check out it?

JiaoXianjun, 2023-06-05:
done

closed 2023-06-05 by JiaoXianjun