open-sdr/openwifi-hw #102: Compiling of FPGA-image failed for "zed_fmcs2-board"

SteffenFri, 2024-01-16:
Hi, 

I followed the last developments of the openwifi project with high interest and I have successfully started the release Notter 1.4.0 by using the "zed_fmcs2-board".

In the following I would like to make some modifications to FPGA-image. At first I have cloned the repository and tried to compile and execute the FPGA-image without any modifications as decribed on your website in the section "Build FPGA". 
By generating the Bitstream using Vivado 2021.1 I get some errors.

At first I get the following output by executing the command "./prepare_adi_lib.sh /tools/Xilinx/":

```
fricke@fricke-ThinkPad-X250:~/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw$ ./prepare_adi_lib.sh /tools/Xilinx/
$XILINX_DIR is found!
++ git submodule init adi-hdl
Submodul 'adi-hdl' (https://github.com/analogdevicesinc/hdl.git) für Pfad 'adi-hdl' in die Konfiguration eingetragen.
++ git submodule update adi-hdl
Klone nach '/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl' ...
Submodul-Pfad: 'adi-hdl': 'f61d9707eb0a62533efd6facab59ab2444da94c9' ausgecheckt
++ cd ./adi-hdl/
++ git reset --hard
HEAD ist jetzt bei f61d9707e ad_fmclidar1_ebz: Update the IO constraints to revB
++ git fetch
++ git checkout 2021_r1
Vorherige Position von HEAD war f61d9707e ad_fmclidar1_ebz: Update the IO constraints to revB
HEAD ist jetzt bei d09fc9207 util_do_ram: Fix Rx path for interrupted transfers
++ git reset --hard 2021_r1
HEAD ist jetzt bei d09fc9207 util_do_ram: Fix Rx path for interrupted transfers
++ source /tools/Xilinx//Vivado/2021.1/settings64.sh
+++ source /tools/Xilinx/DocNav/.settings64-DocNav.sh
++++ '[' -n /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /tools/Xilinx/Vivado/2021.1/.settings64-Vivado.sh
++++ export XILINX_VIVADO=/tools/Xilinx/Vivado/2021.1
++++ XILINX_VIVADO=/tools/Xilinx/Vivado/2021.1
++++ '[' -n /tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /tools/Xilinx/Vitis/2021.1/.settings64-Vitis.sh
++++ export XILINX_VITIS=/tools/Xilinx/Vitis/2021.1
++++ XILINX_VITIS=/tools/Xilinx/Vitis/2021.1
++++ '[' -n /tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /tools/Xilinx/Model_Composer/2021.1/.settings64-Model_Composer.sh
++++ '[' -n /tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/Model_Composer/2021.1/bin:/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
++++ PATH=/tools/Xilinx/Model_Composer/2021.1/bin:/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
+++ source /tools/Xilinx/Vitis_HLS/2021.1/.settings64-Vitis_HLS.sh
++++ export XILINX_HLS=/tools/Xilinx/Vitis_HLS/2021.1
++++ XILINX_HLS=/tools/Xilinx/Vitis_HLS/2021.1
++++ '[' -n /tools/Xilinx/Model_Composer/2021.1/bin:/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2021.1/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2021.1/aietools/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
++++ export PATH=/tools/Xilinx/Vitis_HLS/2021.1/bin:/tools/Xilinx/Model_Composer/2021.1/bin:/tools/Xilinx/Vitis/2021.1/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2021.1/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2021.1/gnu/aarch64/lin/aarch64-none [... cut ...]
create_ip_repo.sh $XILINX_DIR
or
create_ip_repo.sh $XILINX_DIR $IP1_NAME $DEF1 $DEF2 ... $IP2_NAME $DEF1 ...
 -IP_NAME: only xpu/tx_intf/rx_intf/openofdm_tx/openofdm_rx/side_ch are allowed
 -   DEFx: will be "`define IP_NAME_DEFx" in ip_name_pre_def.v for $IP_NAME
 
+ vivado -source ./ip_repo_gen.tcl

****** Vivado v2021.1 (64-bit)
  **** SW Build 3247384 on Thu Jun 10 19:36:07 MDT 2021
  **** IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

start_gui
```

SteffenFri, 2024-01-16:
The program Vivado 2021.1 started. The Tcl Console in the program Vivado has the following output:
```
start_gui
source ./ip_repo_gen.tcl
# exec rm -rf ip_repo
# exec mkdir ip_repo
# exec cp ../../ip/board_def.v ./ip_repo/ -f
# set  fd  [open  "./ip_repo/openwifi_hw_git_rev.v"  w]
# set HASHCODE [exec ../../get_git_rev.sh]
# puts $fd "`define OPENWIFI_HW_GIT_REV (32'h$HASHCODE)"
# close $fd
# set has_side_ch 1 
# set  fd  [open  "./ip_repo/has_side_ch_flag.v"  w]
# if {$has_side_ch > 0} {
#   puts $fd "`define HAS_SIDE_CH 1"
# } else {
#   puts $fd "`define NO_SIDE_CH 1"
# }
# close $fd
# set small_fpga 1
# set  fd  [open  "./ip_repo/fpga_scale.v"  w]
# if {$small_fpga == 1} {
#   puts $fd "`define SIDE_CH_LESS_BRAM 1"
# }
# close $fd
# set NUM_CLK_PER_US 100
# set  fd  [open  "./ip_repo/clock_speed.v"  w]
# puts $fd "`define NUM_CLK_PER_US $NUM_CLK_PER_US"
# if {$small_fpga == 1} {
#   puts $fd "`define SMALL_FPGA 1"
# }
# close $fd
# set grounded_rf_port 0
# set  fd  [open  "./ip_repo/spi_command.v"  w]
# if {$grounded_rf_port == 1} {
#   puts $fd "`define SPI_HIGH 24'hC22001"
#   puts $fd "`define SPI_LOW 24'hC02001"
# } else {
#   puts $fd "`define SPI_HIGH 24'h088A01"
#   puts $fd "`define SPI_LOW 24'h008A01"
# }
# close $fd
# set BOARD_NAME [lindex [split [exec pwd] /] end]
# puts $BOARD_NAME
zed_fmcs2
# set ultra_scale_flag 0
# set part_string xc7z020clg484-1
# set argc 4
# set ip_name openofdm_rx
# if {[file exists ./ip_config/$ip_name\_pre_def.v]==0} {file mkdir ip_config; exec echo "" > ./ip_config/$ip_name\_pre_def.v}
# exec rm -rf project_1
# set current_dir [pwd]
# set argv [list $ultra_scale_flag $current_dir/../../ip/$ip_name $current_dir/ip_repo/$ip_name $BOARD_NAME]
# source ../package_ip_openofdm_rx.tcl
## set ultra_scale_flag [lindex $argv 0]
## set src_dir [lindex $argv 1]
## set ip_dir [lindex $argv 2]
## set BOARD_NAME [lindex $argv 3]
## exec rm -rf project_1
## exec rm -rf $ip_dir
## set current_dir [pwd]
## cd $src_dir/
## set argc 1
## set argv [list $BOARD_NAME]
## source ./openofdm_rx.tcl
### set ARGUMENT1 [lindex $argv 0]
### set ARGUMENT2 [lindex $argv 1]
### set ARGUMENT3 [lindex $argv 2]
### set ARGUMENT4 [lindex $argv 3]
### set ARGUMENT5 [lindex $argv 4]
[... 1175 lines cut ...]
INFO: [IP_Flow 19-4728] Bus Interface 'm00_axis_aclk': Added interface parameter 'ASSOCIATED_BUSIF' with value 'm00_axis'.
INFO: [IP_Flow 19-4728] Bus Interface 's00_axi_aclk': Added interface parameter 'ASSOCIATED_BUSIF' with value 's00_axi'.
INFO: [IP_Flow 19-4728] Bus Interface 's00_axis_aclk': Added interface parameter 'ASSOCIATED_BUSIF' with value 's00_axis'.
INFO: [IP_Flow 19-4728] Bus Interface 'm00_axis_aclk': Added interface parameter 'ASSOCIATED_RESET' with value 'm00_axis_aresetn'.
INFO: [IP_Flow 19-4728] Bus Interface 's00_axi_aclk': Added interface parameter 'ASSOCIATED_RESET' with value 's00_axi_aresetn'.
INFO: [IP_Flow 19-4728] Bus Interface 's00_axis_aclk': Added interface parameter 'ASSOCIATED_RESET' with value 's00_axis_aresetn'.
INFO: [IP_Flow 19-2181] Payment Required is not set for this core.
INFO: [IP_Flow 19-2187] The Product Guide file is missing.
## ipx::unload_core $ip_dir/component.xml
## ipx::edit_ip_in_project -upgrade true -name tmp_edit_project -directory $ip_dir $ip_dir/component.xml
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2021.1/data/ip'.
INFO: [IP_Flow 19-795] Syncing license key meta-data
## update_compile_order -fileset sources_1
## set_property core_revision 2 [ipx::current_core]
## ipx::update_source_project_archive -component [ipx::current_core]
## ipx::create_xgui_files [ipx::current_core]
## ipx::update_checksums [ipx::current_core]
## ipx::save_core [ipx::current_core]
## ipx::move_temp_component_back -component [ipx::current_core]
## close_project -delete
## close_project -delete
# exec rm -rf ./ip_repo/$ip_name/xgui
```

SteffenFri, 2024-01-16:
In the end I would like to generate the bitstream using the program Vivado. While running this command the IP Core generation and the synthesis failed and I get the following output in the tab "Log": 

```
*** Running vivado
    with args -log system_top.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source system_top.tcl

****** Vivado v2021.1 (64-bit)
  **** SW Build 3247384 on Thu Jun 10 19:36:07 MDT 2021
  **** IP Build 3246043 on Fri Jun 11 00:30:35 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source system_top.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl/library'.
INFO: [IP_Flow 19-1700] Loaded user IP repository '/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/ip_repo'.
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2021.1/data/ip'.
WARNING: [BD 41-1661] One or more IPs have been locked in the design 'system.bd'. Please run report_ip_status for more details and recommendations on how to fix this issue.
List of locked IPs:
system_openofdm_rx_0_0

Command: synth_design -top system_top -part xc7z020clg484-1 -directive PerformanceOptimized -retiming -fsm_extraction one_hot -keep_equivalent_registers -resource_sharing off -no_lc -shreg_min_size 5
Starting synth_design
WARNING: [Vivado_Tcl 4-393] The 'Synthesis' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system.bd

WARNING: [Vivado_Tcl 4-393] The 'Implementation' target of the following IPs are stale, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system.bd

Attempting to get a license for feature 'Synthesis' and/or device 'xc7z020'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7z020'
INFO: [Device 21-403] Loading part xc7z020clg484-1
INFO: [Synth 8-7079] Multithreading enabled for synth_design using a maximum of 4 processes.
INFO: [Synth 8-7078] Launching helper process for spawning children vivado processes
INFO: [Synth 8-7075] Helper process launched with PID 1049
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:04 ; elapsed = 00:00:05 . Memory (MB): peak = 2399.660 ; gain = 11.793 ; free physical = 123 ; free virtual = 1365
---------------------------------------------------------------------------------
INFO: [Synth 8-6157] synthesizing module 'system_top' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system_top.v:40]
INFO: [Synth 8-6157] synthesizing module 'ad_iobuf' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl/library/common/ad_iobuf.v:38]
	Parameter DATA_WIDTH bound to: 49 - type: integer 
INFO: [Synth 8-6155] done synthesizing module 'ad_iobuf' (1#1) [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl/library/common/ad_iobuf.v:38]
INFO: [Synth 8-6157] synthesizing module 'ad_iobuf__parameterized0' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl/library/common/ad_iobuf.v:38]
	Parameter DATA_WIDTH bound to: 2 - type: integer 
INFO: [Synth 8-6155] done synthesizing module 'ad_iobuf__parameterized0' (1#1) [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/adi-hdl/library/common/ad_iobuf.v:38]
INFO: [Synth 8-6157] synthesizing module 'system_wrapper' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system_wrapper.v:14]
INFO: [Synth 8-6157] synthesizing module 'IOBUF' [/tools/Xilinx/Vivado/2021.1/scripts/rt/data/unisim_comp.v:56188]
INFO: [Synth 8-6155] done synthesizing module 'IOBUF' (2#1) [/tools/Xilinx/Vivado/2021.1/scripts/rt/data/unisim_comp.v:56188]
INFO: [Synth 8-6157] synthesizing module 'system' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/synth/system.v:5940]
ERROR: [Synth 8-439] module 'system_GND_1_0' not found [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/synth/system.v:6502]
ERROR: [Synth 8-6156] failed synthesizing module 'system' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/synth/system.v:5940]
ERROR: [Synth 8-6156] failed synthesizing module 'system_wrapper' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system_wrapper.v:14]
ERROR: [Synth 8-6156] failed synthesizing module 'system_top' [/home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/system_top.v:40]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:06 ; elapsed = 00:00:08 . Memory (MB): peak = 2448.535 ; gain = 60.668 ; free physical = 99 ; free virtual = 888
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
19 Infos, 3 Warnings, 0 Critical Warnings and 5 Errors encountered.
[... cut ...]
CRITICAL WARNING: [BD 41-1938] The current design is out-of-date/locked, so it is best to update before calling write_bd_tcl, however, a Tcl script is still generated for the current design.
Generated Block Design Tcl file /home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/hw_handoff/system_bd.tcl
Generated Hardware Definition File /home/fricke/Dokumente/OpenwifiHardwareImage_16_01_2024/openwifi-hw/boards/zed_fmcs2/src/synth/system.hwdef
ERROR: [Vivado 12-4756] Launch of runs aborted due to earlier errors while preparing sub-designs for run execution.
```

JiaoXianjun, 2024-01-19:
Did you check the "Pre-conditions:" in our README?

closed 2024-02-29 by SteffenFri

chenmo234, 2024-05-28:
I encountered the same issue as you. Can we discuss it? My contact information is c2385405729.