open-sdr/openwifi-hw #108: Requesting assistance to confirm that what I have discovered is not a bug.

showna, 2024-07-19:
Hello Dr. Jiao!Recently, I have been studying the Verilog code of openwifi-hw and encountered an issue. I'm unsure whether it's a bug or by design, and I hope to get clarification. In the file dot11_tx.v, at line 255, there is a process where when plcp_bit_cnt equals 22, bram_addr is set to 1. What is the purpose of this logic? According to this logic, wouldn't the PLCP data read at line 118 be potentially problematic?

mmehari, 2024-07-19:
You have to look the way PLCP header is encoded in a packet with in the driver but 
L-SIG + PKT_TYPE is encoded in the 1st 64bit word of bram followed by HT-SIG on the 2nd 64bit word.

Given 2 clock pulses required for bram access, we retrieve HT-SIG after finish reading L-SIG, which is when plcp_bit_cnt equals 22.

showna, 2024-07-20:
> You have to look the way PLCP header is encoded in a packet with in the driver but L-SIG + PKT_TYPE is encoded in the 1st 64bit word of bram followed by HT-SIG on the 2nd 64bit word.
>
> Given 2 clock [...]

Since the L-SIG has a length of 24 bits, would setting the BRAM address bus to the next 64-bit word when plcp_cnt equals 22 possibly lead to an issue with the last bit of the L-SIG content being read incorrectly on line 151 of the file?

mmehari, 2024-07-20:
It wouldn't and you can see it yourself by simulating dot11_tx_tb.v

showna, 2024-07-21:
> It wouldn't and you can see it yourself by simulating dot11_tx_tb.v

Shouldn't the `bram_addr` be set to 1 when `plcp_bit_cnt` equals 23? Although the result might be correct, I'm struggling to understand this particular point, and it's confusing me.

closed 2024-07-22 by showna