open-sdr/openwifi-hw #110: Why doesn't OFDM_TX use the xfft IP ？

clark-terry, 2024-08-20:
Hello Dr. Jiao
   ofdm_ rx used  xilinx xfft IP, xfft ip  datasheet's introduction  can also be used for IDFT. Why doesn't OFDM_TX use the xfft IP, but instead has to write its own? Is there something about the xfft IP that doesn't meet the requirements?

JiaoXianjun, 2024-08-20:
Please fill in our issue template.

clark-terry, 2024-08-21:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
  yes
1. Our image is used directly or you build your own image?
  I used  your image.
2. What is your own modification?
  I didn't make any changes
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
 the master branch.
4. Board/hardware type
antsdr
5. WiFi channel number
 1t1r
6. Steps to reproduce the issue, and the related error message, screenshot, etc
NA
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
NA
8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
NA
9. Any other thing we need to know for helping you better? 
ofdm_ rx used xilinx xfft IP, xfft ip datasheet's introduction can also be used for IDFT. Why doesn't OFDM_TX use the xfft IP, but instead has to write its own? Is there something about the xfft IP that doesn't meet the requirements?

mmehari, 2024-08-23:
I would say a **lucky coincidence** during the development phase and I happened to know the open source FFT module (from Gisselquist Technology: GT) before hand.

Although I haven't tested it myself, I am sure the xfft IP should work just fine.

closed 2024-08-26 by clark-terry