open-sdr/openwifi-hw #112: Some questions about the code.

showna, 2024-08-22:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
I did not introduce myself via email.
1. Our image is used directly or you build your own image?
Not used directly; just for learning.
2. What is your own modification?
 Just for learning.
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
OS:ubuntu 1604, Vivado 2021.1,openwifi-hw/master.
4. Board/hardware type
NONE
5. WiFi channel number
NONE
6. Steps to reproduce the issue, and the related error message, screenshot, etc
NONE
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
NONE
8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
NONE
9. Any other thing we need to know for helping you better? 

         if(fifo_turn == PKT_FIFO) begin
            if(pkt_iq_sent[5:0] == 6'b111111 && |pkt_iq_sent[15:8] == 1)
                fifo_turn <= CP_FIFO;
            pkt_iq_sent   <= pkt_iq_sent + 1;
        end else if(fifo_turn == CP_FIFO) begin
            if((pkt_iq_sent < 640 && CP_iq_sent[3:0] == 4'b1111) || (S_GI == 0 && CP_iq_sent[3:0] == 4'b1111) || (S_GI == 1 && pkt_iq_sent >= 640 && CP_iq_sent[2:0] == 3'b111))
                fifo_turn <= PKT_FIFO;
            CP_iq_sent    <= CP_iq_sent + 1;
        end

According to the logic of this code segment: within the range of 128 to 32767, insert 16 CPs every 64 data sampling points. I would like to know the origin of this logic. Why does it start from "128"? What do "32767" and "640" refer to? Thank you for providing answers.

mmehari, 2024-08-22:
I don't fully understand your question, but the code segment you posted is used to sequentially count the number of IQ samples that are used both for packet and cyclic prefix (CP).

Furthermore, the values **128** and **32767** are not indicated in the code segment but the value **640** refers to the number of data IQ samples (excluding CP) up to the first data OFDM symbol of an 802.11n packet (L-STF + L-LTF + L-SIG + HT-STF + HT-LTF) (160 + 160 + 64 + 128 + 64 + 64)

showna, 2024-08-23:
> I don't fully understand your question, but the code segment you posted is used to sequentially count the number of IQ samples that are used both for packet and cyclic prefix (CP).
> 
> Furthermore, the values **128** and **32767** are not indicated in the code segment but the value **640** refers to the number of data IQ samples (excluding CP) up to the first data OFDM symbol of an 802.11n packet (L-STF + L-LTF + L-SIG + HT-SIG + HT-STF + HT-LTF) (160 + 160 + 64 + 128 + 64 + 64)

My apologies for the confusion, it should be the values **255** and **65536**.
When the condition **|pkt_iq_sent[15:8] == 1** is met, the value range of pkt_iq_sent is from **255** to **65535**.

As for your additional question, in the dot11_tx.v code, the **service field** does not participate in the **alignment** operation of the data field. Wouldn't this approach cause any issues?

mmehari, 2024-08-23:
> My apologies for the confusion, it should be the values 256 and 65536.
When the condition |pkt_iq_sent[15:8] == 1 is met, the value range of pkt_iq_sent is from 256 to 65535.
What do 256 and 65535 refer to?

You have to consider the entire line `if(pkt_iq_sent[5:0] == 6'b111111 && |pkt_iq_sent[15:8] == 1)` and instead of **256**, the value becomes **64+256 = 320**, which is the start of the first CP IQ sample residing inside **L-SIG** field. And the search runs until the end of a packet of maximum size **65535**.

> In the code segment, what do the values 480 and 240 respectively refer to?

The numbers **240** and **480** are specific to 802.11n packets such that **480** refers to the number of IQ samples present until mid-point HT-SIG and **240** is the number of IQ samples past **480** to reach the first OFDM symbol. 

As a side note, if you are curious to understand **dot11_tx**, the best approach will be simulating it inside Vivado using the provided test vectors.

showna, 2024-08-24:
Here's my suggestion: Could you consider adding comments or using macros to replace these constant numbers where they are critical? This would enhance the code readability.

mmehari, 2024-08-24:
Indeed that will enhance code readability but unfortunately I am not active on the project and I can't say what I can't do.

closed 2024-10-11 by showna

ll550, 2024-10-15:
thx for discussion，quite insightful