open-sdr/openwifi-hw #116: Need Guidance on High-Level Basis for Verilog Implementation

Tagabey, 2024-12-16:
Hi OpenWiFi Team,

I’m an undergraduate student in Electrical and Electronics Engineering. My team and I are using OpenWiFi-HW for our graduation project on wireless communication.

We’re finding it challenging to understand the physical layer without knowing the high-level algorithms behind the Verilog code. Are there any high-level verification codes or resources, like blogs, tutorials, or textbooks, that you can recommend to help us out?

mmehari, 2024-12-17:
Part of the receive path in the openwifi baseband is based on OpenOFDM (Synthesizable, Modular Verilog Implementation of 802.11 OFDM Decoder).
You can read the documentation from https://openofdm.readthedocs.io/en/latest/

Tagabey, 2025-04-18:
Thank you for your response.
The OpenOFDM wiki explains the receiver modules, but I also need to understand the transmitter (TX) side. Additionally, I’d like to learn about the workflow for comparing different modules with golden models.
I’m a final-year engineering student, and I’m looking for advice on how to learn the methodology for designing and working with communication systems.

mmehari, 2025-04-19:
For the transmitter side, I am afraid you have to rely on the source code. You can simulate the transmitter module with a given test vector (e.g. [tx_intf.mem](https://github.com/open-sdr/openwifi-hw/blob/master/ip/openofdm_tx/unit_test/test_vec/tx_intf.mem), and understand its operation. From complexity point of view, transmitter is way less complex than the receiver, since the receiver has to handle the imperfections/distortions of the channel.

As to your other questions, I am not working on the project anymore and I can't help you on that.

ViruKM, 2025-04-28:
Hi team 
I am using openwifi-hw, try to build fpga block design and need to change in SPI.V, but i am unable to open block design after .openwifi.tcl command runs.
can you help me how to solve the this problem, i am using vivado 2021.1 and ubuntu 20.4LTS 64 bit.

weiliu1011, 2025-04-28:
> Hi team I am using openwifi-hw, try to build fpga block design and need to change in SPI.V, but i am unable to open block design after .openwifi.tcl command runs. can you help me how to solve the this problem, i am using vivado 2021.1 and ubuntu 20.4LTS 64 bit.

Spi.v is in the XPU up core, after modify this file u need to recreate IP repo, and follow all steps afterwards in our readme

ViruKM, 2025-04-29:
Hi,
 Thank you for your response, 
I followed all the steps given in readme. but i am getting error while opening vivado tool. unable to open respective board openwifi.tcl file in the vivado

ViruKM, 2025-04-29:
Hi,
after follow all the steps. i am facing theses errors so kindly tell me how to resolve this errors

[image] 
I am facing theses errors

weiliu1011, 2025-04-29:
Based on your screenshot, you did not run the script in the correct directory, that is why the IP cores are all not found. The correct way is (as shown in readme)
cd openwifi-hw/boards/$BOARD_NAME/
../create_ip_repo.sh $XILINX_DIR

ViruKM, 2025-04-29:
I used all below steps 

export XILINX_DIR=your_Xilinx_install_directory

(Example: export XILINX_DIR=/opt/Xilinx. The Xilinx directory should include sth like: Downloads, SDK, Vivado, xic)
cd openwifi_hw ./prepare_adi_lib.sh $XILINX_DIR

    Prepare Analgo Devices specific ip (only run once for each board you have):

export BOARD_NAME=your_board_name
(Example: export BOARD_NAME=zc706_fmcs2)

./prepare_adi_board_ip.sh $XILINX_DIR $BOARD_NAME
(Don't need to wait till the building end. When you see "Building ABCD project [...", you can stop it.)

    Get the openofdm_rx into ip directory (only run once after openofdm is udpated):

./get_ip_openofdm_rx.sh

    Generate ip_repo for the top level FPGA project (will take a while):

cd openwifi-hw/boards/$BOARD_NAME/
../create_ip_repo.sh $XILINX_DIR

after this i gave source ./openwifi.tcl ( full path of board folder).

weiliu1011, 2025-04-29:
you don't need to source openwifi.tcl, normally it is called by create_ip_repo, after ../create_ip_repo.sh you should see the project generated.

ViruKM, 2025-04-29:
steps i follows 
1  export XILINX_DIR=your_Xilinx_install_directory(tools/XIlinx)
2  openwifi-hw$ ./prepare_adi_lib.sh $XILINX_DIR
3 openwif-hwi $export BOARD_NAME=your_board_name (antsdre200)
4  openwifi$./get_ip_openofdm_rx.sh
5 openwifi-hw$ cd boards/$BOARD_NAME
6 openwifi-hw/boards/antsd_e200$./create_ip_repo.sh $XILINX_DIR
7 source  /openwifi-hw/boards/antsd_e200/openwifi.tcl     
after this am getting above ip errors

ViruKM, 2025-04-29:
> you don't need to source openwifi.tcl, normally it is called by create_ip_repo, after ../create_ip_repo.sh you should see the project generated.

this open  openofdm_rx as a top file and unable to open block design for this.

please help me to resolve this soon

ViruKM, 2025-04-29:
[image]  

 if, I am doing source  ./openwifi.tcl openofdm.tcl will open as  i attached and unable to open block diagram, no ip and sytem_top not there in this design.

ViruKM, 2025-04-30:
[image]

as you told, i did but synthesis errors. please guide me how to solve it.